carla-simulator/carla #1856: [0.9.5] error when running the "automatic_control.py"

fty0724, 2019-07-08:
hello,everyone. I meet an error when running the script"automatic_control.py". And the error is showing below:

> Hello from the pygame community. https://www.pygame.org/contribute.html
> Traceback (most recent call last):
>   File "automatic_control.py", line 120, in <module>
>     from agents.navigation.roaming_agent import RoamingAgent
> ImportError: No module named agents.navigation.roaming_agent

I have tried the way just given by this link [https://github.com/carla-simulator/scenario_runner/issues/35](url) . However, changing the PYTHONPATH has no help.

I also tried the way [https://github.com/carla-simulator/carla/issues/1211](url). the package is imported successfully, but the error has no change.

Anyone can help me?

jackbart94, 2019-07-08:
Hi, there is an import that doesn't go through, maybe because the folder structure was changed.

A possible fix is to modify line 82 to:
sys.path.append(os.path.dirname(os.path.dirname(os.path.abspath(__file__))) + '/carla')

Or you can put the agents folder (which is a subfolder of carla) in the example one, and it should work.

fty0724, 2019-07-10:
Thanks, I put the agents folder in the example and it really works!

closed 2019-07-10 by fty0724