carla-simulator/carla #1869: Is there any way for a camera to record video instead of take pictures?

programmer6288, 2019-07-09:
Hello,

Pretty much what the title says. Right now I am using ffmpeg to turn the images into an mp4, but is there any method in carla to do this? I am noticing that there are some periodic frame drops.

Thank you

nsubiron, 2019-07-09:
Hi @programmer6288,

Use the [synchronous mode](https://carla.readthedocs.io/en/latest/configuring_the_simulation/#synchronous-mode) to avoid frame drops.

programmer6288, 2019-07-11:
Hello,

I am currently unclear on how to run synchronous mode. When I add this code:

> settings = world.get_settings()
settings.synchronous_mode = True
world.apply_settings(settings)

to the script that I am running, the scripts fails due to collision, and the simulation becomes unresponsive (fades to black). What am I doing wrong here?

Thank you

p-schulz, 2019-07-15:
Hi,

when using the python client, have a look at **imageio**. It has the ability to pipe images directly to the ffmpeg encoding. This helped me as I encountered frame drops when writing single images to disk. You'll have the same framerate as in simulation without recording and instant an mp4 file.

closed 2019-09-20 by stale[bot]