carla-simulator/carla #1872: Commands of Setup.bat not found after make launch

seffylmz, 2019-07-11:
I try to build Carla 0.9.5.
After running `make launch` I get this error:

> C:\Users\user\carla>make launch
C:/Users/user/carla/Util/BuildTools/Setup.bat: line 1: @echo: command not found
C:/Users/user/carla/Util/BuildTools/Setup.bat: line 2: setlocal: command not found
C:/Users/user/carla/Util/BuildTools/Setup.bat: line 4: rem: command not found
C:/Users/user/carla/Util/BuildTools/Setup.bat: line 5: syntax error near unexpected token `('
C:/Users/user/carla/Util/BuildTools/Setup.bat: line 5: `rem rpclib, gtest and boost libraries for CARLA (carla.org).'
make: *** [setup] Fehler 258

closed 2019-09-16 by stale[bot]

roser-br, 2020-01-28:
Hi @seffylmz! 
Were you able to solve this issue? I'm facing the same situation at the moment. 
Thanks in advance :)

seffylmz, 2020-01-28:
> Hi @seffylmz!
> Were you able to solve this issue? I'm facing the same situation at the moment.
> Thanks in advance :)

Unfortunately, I didn‘t found a solution. But after resetting the Windows OS to factory settings it worked...