carla-simulator/carla #1873: Inconsistent RuntimeError because of collision at spawn position

zeinramadan, 2019-07-11:
I'm trying to spawn 2 vehicles at DIFFERENT coordinates and have them move along points that are read from a csv file. One of my python scripts I have runs 1 in 3 tries the other doesn't run at all, it just throws a RuntimeError saying that the spawn failed because of collision at spawn position which makes no sense to me since the initial coordinates are reasonably far away from one another. Any suggestions as to what might be causing the issue?

karthi0804, 2019-07-17:
Hi, How do you specify the location for spawning the vehicle? I solved this by finding the nearest waypoint and spawning according to that location.
```object_veh = world.spawn_actor(obj_blueprint, map.get_waypoint(object_transform.location).transform)```

I guess the reason is the difference in Z value which makes Carla think as a collision with the ground.

zeinramadan, 2019-07-22:
> Hi, How do you specify the location for spawning the vehicle? I solved this by finding the nearest waypoint and spawning according to that location.
> `object_veh = world.spawn_actor(obj_blueprint, [...]

zeinramadan, 2019-07-22:
Solved the issue, apparently if I abruptly end the script the actors don't get properly destroyed so running the script again would spawn the actors on an already existing actor. Just had to restart UE.

closed 2019-07-22 by zeinramadan

dhdiagram4011, 2022-03-10:
@karthi0804 

custom map carla import --> linux command line python lidar_to_camera.py play
RuntimeError: Spawn failed because of collision at spawn position

Kenneth71-P, 2022-04-14:
I have the same problem. Can you tell me how can i solve this?
[image]
[image]