carla-simulator/carla #1875: vehicle.set_location() problem

Zealoft, 2019-07-12:
- CARLA version(s): 0.9.4 , overnight 2019-03-19 and probably others

- Platform: Ubuntu 16.04

Recently I have met a problem when using world.vehicle.set_location() function. Concretely, I randomly pick a location in the map(all on the road), initialise a client and use the world.vehicle.set_location() method to set the initial location. However, when I try to get the current location using get_location() method, there is always a random bias between the actual location and the location I set(in fact I have tried time.sleep() and wait_for_tick() method but none worked). Below is a chart about the bias in Carla's map Town 03.
[image]
Has anybody met similar problem and can help me work it out? Thank you very much!

PimenovAlexander, 2019-07-17:
I can confirm this issue exists. It seriously affects reproducibility.
1. It can be reproduced for all maps
2. It seem to affect real image - actual point of view is also shifted.
3. This shift seem to be higher if the car has non-zero pitch or roll rotation. This makes me think physics affects the coordinates. But it also can be reproduced with  vehicle.set_simulate_physics(False). 
4. If all the objects in the scene are positioned in the same way, this random shift seen to repeat itself.
5. I would try to create a snippet to reproduce the issue

I consider this bug to be extremely critical, it affects all scene geometry, reproducibility and all attempts to compute optical flow.

PimenovAlexander, 2019-07-17:
Here is a piece of code that shows the issue
`#!/usr/bin/env python

import glob
import os
import sys

sys.path.append("../../carla9.6/PythonAPI/carla/dist/carla-0.9.6-py2.7-linux-x86_64.egg")
import carla
    
def camera_callback_dbg(image, outpath) :               
    global frame_number
    name="000_" + str(frame_number) + '.png'
    print name
    image.save_to_disk(name)
    print(image.transform)

actor_list = []
    
if __name__ == '__main__':        
    global frame_number
    frame_number=0
    
    print ("Try to reproduce")
    client = carla.Client("localhost", 2000)
    client.set_timeout(8.0)
    world = client.get_world()  
    
    try:       
        settings = world.get_settings()
        settings.synchronous_mode = True
        settings.fixed_delta_seconds = 0.1;
        world.apply_settings(settings)
        print ("Switched to sync mode")
        
        world.set_weather(carla.WeatherParameters(cloudyness=0, precipitation=0, sun_altitude_angle=35))

        blueprint_library = world.get_blueprint_library()
        blueprints_v = blueprint_library.filter('vehicle.*')
        bp = blueprints_v[0]
        
        spawn_point =  carla.Transform(carla.Location(x=0.0, y=0.0, z=100.0), carla.Rotation(pitch=2.0, yaw=0, roll=3)) 
        
        print (bp)
        vehicle = world.try_spawn_actor(bp, spawn_point)
        vehicle.set_simulate_physics(False)        
        actor_list .append(vehicle)
        
        camera_bp = blueprint_library.find("sensor.camera.rgb")
        camera_bp.set_attribute('image_size_x', str(4000))
        camera_bp.set_attribute('image_size_y', str(4000))
        camera_bp.set_attribute('fov', str(60))

        loc = carla.Location(x=0,y=0, z=2)
        rot = carla.Rotation(pitch=0, yaw=0, roll=0)
        camera_transform = carla.Transform(loc, rot)
        camera = world.spawn_actor(camera_bp, camera_transform, attach_to=vehicle)
        actor_list.append(camera)
   
        print ("Registering callback")        
        camera.listen(lambda t: (camera_callback_dbg (t, ".")))
        
        print ("Making debug draw")
        
        begin = carla.Location(x=60.0, y= -20.0, z=102.0)
        end =   carla.Location(x=60.0, y=  20.0, z=102.0)        
        world.debug.draw_line(begin, end, color=carla.Color(255,0,0), life_time=4)
        
        begin = carla.Location(x=60.0, y= 0.0, z=122.0)
        end =   carla.Location(x=60.0, y= 0.0, z= 82.0)        
        world.debug.draw_line(begin, end, color=carla.Color(255,0,0), life_time=4)

        begin = carla.Location(x=160.0, y= -20.0, z=102.0)
        end =   carla.Location(x=160.0, y=  20.0, z=102.0)        
        world.debug.draw_line(begin, end, color=carla.Color(0,255,0), life_time=4)
        begin = carla.Location(x=160.0, y= 0.0, z=122.0)
        end =   carla.Location(x=160.0, y= 0.0, z= 82.0)        
        world.debug.draw_line(begin, end, color=carla.Color(0,255,0), life_time=4)
                

        frame_number = 0
        world.tick();
        frame_number = 1        
        world.tick();
        frame_number = 2        
        world.tick();
        frame_number = 3        
        world.tick();
        frame_number = 4        
        world.tick();
        frame_number = 5        
        world.tick();        
        print ("Finished")         
        
    except Exception as error: 
        print('Clean up')
        print (error)
    finally:    
        print (actor_list)        
        for actor in actor_list:
            actor.destroy()
        
        world.tick();
        
    settings = world.get_settings()
    settings.synchronous_mode = False
    world.apply_settings(settings)   
  `

Zealoft, 2019-07-18:
> I can confirm this issue exists. It seriously affects reproducibility.
> 
> 1. It can still be seen in 0.9.5 and 0.9.6 for Ubuntu 18.10
> 2. It can be reproduced for all maps
> 3. It seem to affect real image - actual point of view is also shifted.
> 4. This shift seem to be higher if the car has non-zero pitch or roll rotation. This makes me think physics affects the coordinates. But it also can be reproduced with  vehicle.set_simulate_physics(False).
> 5. If all the objects in the scene are positioned in the same way, after the script restart this random shift seem to repeat itself. So it is fully determined by the scene.
> 6. I would try to create a snippet to reproduce the issue
> 
> I consider this bug to be extremely critical, it affects all scene geometry, reproducibility and all attempts to compute optical flow.

Thanks for the reply! So it seems to be a critical bug in the current Carla version. So are there other ways to set the vehicle position manually or will this bug be fixed soon? Thanks a lot~

closed 2019-09-23 by stale[bot]