carla-simulator/carla #1876: What's the function relationship between throttle and velocity, steer and heading angle?

linxigjs, 2019-07-12:
My purpose is to control the vehicle's velocity(both it's value and direction) by acceleration and the angle of the front wheel, so what's the relationship between them? What's the calculation model? Any code in Carla, or reference articles are appreciate. Thanks.

xiaoyang666, 2019-07-16:
The specific model should be placed in UE4 according to a vehicle model from NVIDIA. I only found some engine and transmission parameters.

linxigjs, 2019-07-18:
> The specific model should be placed in UE4 according to a vehicle model from NVIDIA. I only found some engine and transmission parameters.
Yes, thanks. I also found the vehicle and wheel model parameters, torque curve as well. But i'm confused why they set this way and what's the relationship between them.

sheelabhadra, 2019-09-18:
Hi @linxigjs, could you please let me know the way to obtain the current wheel angle?

linxigjs, 2019-09-20:
Hi @sheelabhadra , If you want to know the front wheel steer angle, you may read the code in carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Vehicle/CarlaWheeledVehicle.cpp line 112, which is 
float ACarlaWheeledVehicle::GetMaximumSteerAngle() const
{
  const auto &Wheels = GetVehicleMovementComponent()->Wheels;
  check(Wheels.Num() > 0);
  const auto *FrontWheel = Wheels[0];
  check(FrontWheel != nullptr);
  return FrontWheel->SteerAngle;
}
Hope it will help you.

closed 2019-11-26 by stale[bot]