carla-simulator/carla #1877: generate_map.py don't appear

eloy96vg, 2019-07-12:
Hi!

In the latest commit (29f6acb) don't appear the file generate_map.py, that is the easiest way to upload RoadRunner maps. Why it does not show up? Is there any new way to upload these maps?

Thanks.

nsubiron, 2019-07-12:
Hi @eloy96vg,

We'll be announcing the new system in the coming release, but meanwhile master branch already has the changes, the new documentation is at [Creating standalone asset packages for distribution](https://carla.readthedocs.io/en/latest/asset_packages_for_dist/).

eloy96vg, 2019-07-24:
Hi @nsubiron ,

There was some way to recover generate_map.py, because I am doing a job with carla and I need to import maps. With this command (generate_map.py) I could import maps without problems, however, with make import I have the problem that I expose in this comment (https://github.com/carla-simulator/carla/issues/1886#issuecomment-512986988).

A greeting.

closed 2019-09-29 by stale[bot]

Sultan91, 2023-10-15:
same for me