carla-simulator/carla #1878: please help "clang 7 is required, but it's not installed"

xiahvc, 2019-07-12:
i follow the doc "how to build on linux" in tutorial page

but when i reach to last section "make launch" it occured this message "clang 7 is required, but it's not installed"

however i can't install clang 7 because i don't know ubuntu well

please help what can i do?

OmarElzaafaranyBrightskies, 2019-07-17:
Please, tell me if you find a solution. I am frozen at the same stage.

asharalam11, 2019-07-17:
I hope this helps, I spent over a day but finally got a solution:

If you get an error related to "Clang 7.0 not installed" while you are installing Carla; go to https://apt.llvm.org/ and install Clang 7 using the command: apt-get install clang-7 clang-tools-7 clang-7-doc libclang-common-7-dev libclang-7-dev libclang1-7 clang-format-7 python-clang-7

asharalam11, 2019-07-17:
> Please, tell me if you find a solution. I am frozen at the same stage.

I hope this helps, I spent over a day but finally got a solution:

If you get an error related to "Clang 7.0 not installed" while you are installing Carla; go to https://apt.llvm.org/ and install Clang 7 using the command: apt-get install clang-7 clang-tools-7 clang-7-doc libclang-common-7-dev libclang-7-dev libclang1-7 clang-format-7 python-clang-7

OmarElzaafaranyBrightskies, 2019-07-19:
Yes, it worked thanks very much. I realized that my firewall was stopping this package "Clang-7" from being downloaded.
while I proceed with the setup: make PythonAPI 
It outputs this error :

clang-6.0 -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -Wdate-time -D_FORTIFY_SOURCE=2 -g -fdebug-prefix-map=/build/python2.7-MW0004/python2.7-2.7.15=. -fstack-protector-strong -Wformat -Werror=format-security -fPIC -Idependencies/include -I/usr/lib/gcc/x86_64-redhat-linux/7/include -I/usr/include/python2.7 -c source/libcarla/libcarla.cpp -o build/temp.linux-x86_64-2.7/source/libcarla/libcarla.o -isystem dependencies/include/system -fPIC -std=c++14 -Werror -Wall -Wextra -Wpedantic -Wno-self-assign-overloaded -Wdeprecated -Wno-shadow -Wuninitialized -Wunreachable-code -Wpessimizing-move -Wold-style-cast -Wnull-dereference -Wduplicate-enum -Wnon-virtual-dtor -Wheader-hygiene -Wconversion -Wfloat-overflow-conversion -DBOOST_ERROR_CODE_HEADER_ONLY -DLIBCARLA_WITH_PYTHON_SUPPORT -DLIBCARLA_IMAGE_WITH_PNG_SUPPORT=true
error: unknown warning option '-Wno-self-assign-overloaded'; did you mean
     '-Wno-self-assign-field'? [-Werror,-Wunknown-warning-option]
error: command 'clang-6.0' failed with exit status 1
Util/BuildTools/Linux.mk:80: recipe for target 'PythonAPI' failed
make: *** [PythonAPI] Error 1

Have you passed by it? 
Thanks very much for your time and efforts.

asharalam11, 2019-07-19:
Did it work for you or are you still stuck? Asking so I can close this issue?

OmarElzaafaranyBrightskies, 2019-07-19:
"clang 7 is required, but it's not installed" is done for me.
You can close the issue if you wish.
Thanks

asharalam11, 2019-07-19:
Glad that I could help :)

closed 2019-09-24 by stale[bot]