carla-simulator/carla #1881: Errors when running synchronous mode

programmer6288, 2019-07-12:
Hello,

I am currently unclear on how to format my code so that I can run it in synchronous mode.

When running synchronous mode, I understand that you must change the world settings to have a fixed time step and be in sync mode. What I don't understand is what must go inside the while True loop. 

How would I write a script that spawns a vehicle, sets it into autopilot, places a camera on it, and stores the images in synchronous mode?

Thank you.

eduardohenriquearnold, 2019-07-31:
I have also been experiencing this problem. The [tutorial on  sync mode](https://github.com/carla-simulator/carla/blob/master/PythonAPI/examples/synchronous_mode.py) spawn the actors before activating the synchronous mode. I required to spawn actors after activating the sync mode. Any ideas on this?

closed 2019-10-06 by stale[bot]

robrivas, 2020-07-19:
I also have an issue with sync mode, pygame crashes immediately as the collision sensor is trying to be listened. I'm using the PythonAPI example file "synchronous_mode", putting a blueprint for the col. sensor, and trying to capture the data inside the while true loop.