carla-simulator/carla #1886: Error with 'make import'

eloy96vg, 2019-07-15:
Hi!

I'm trying import maps from Roadrunner with the new command "make import" following the instructions but I'm having some problems. I have the folders and the JSON file as shown in the figures below.

[image]

[image]

[image]

But when i run the "make import" appear this error.

[image]

I'm waiting for your help, thanks in advance.

manishthani, 2019-07-17:
Hi @eloy96vg ,
Seems like you are missing `props: []` in your `Package1.json` file, which is an empty JSON array **only if you have no props to import**.
So, basically, try adding that to your file:
```
{
     maps: [
     // Your current maps
     ],
     props: []
} 
```
Please, let me know if it worked.

eloy96vg, 2019-07-18:
Thanks @manishthani !
That solved the problem, but when i import the maps RoutePlanners and Spawn points are created, 
but the nodes are not created on the map. In fact, as you can see in the image, a single node has been created. Would you know why this happens? I would be very helpful.

[image]

Thanks and best regards

manishthani, 2019-07-22:
Hi @eloy96vg,
Maybe this could happen if the OpenDrive `xodr` file is not correct, could you please check it again and try opening it with the [OpenDrive visualizer](http://www.opendrive.org/download.html)?

eloy96vg, 2019-07-22:
Hi @manishthani,
I'm tryiang to install OpenDrive visualizer in Ubuntu but i cant (It says "There are no applications available to open << odrViewer.1.9.1 >>"), do you know how can i download it?
On the other hand, comment that with the old form available (generate_map.py) the map was created correctly. The problem has been with make import, where the nodes are not created.

closed 2019-09-02 by eloy96vg