carla-simulator/carla #1889: Contribute client_bounding_boxes.py Module.

harderthan, 2019-07-16:
Following this issue, #1743 

Hello. Maintainers.

I also need the module, which @qingmingjie told, for trainning my custom object detection model.
If there is no module to do like that in this Carla repository, could I develop and PR it?

I'm considering to develop the module like following. I will remove not necessary bounding boxes of images which we can get from client_bounding_boxes.py.

Maybe, using segmentation information on that's images, I can distinguish with visible object or not.
If there is some problems to solve this, could you advice better options to me?

Soolek, 2019-07-24:
Hi,
It depends on your requirements. The way we are doing it is by fusing Lidar points with bounding boxes and returning vehicles that are less than x% occluded.

harderthan, 2019-07-26:
@Soolek, Hi.

Then, Are there other requirements for the load map of this repository? 
I want to develop more general module for me and other people.

If you don't have other requirements, I will contribute the result after development.

closed 2019-10-01 by stale[bot]