carla-simulator/carla #1892: 0.9.6 docker container: xdg-user-dir not found

fpasch, 2019-07-16:
When executing 0.9.6 docker with
```
$ nvidia-docker run -it --rm  carlasim/carla:0.9.6 ./CarlaUE4.sh
```

The following warning is reported:
```
4.22.1-0+++UE4+Release-4.22 517 0
Disabling core dumps.
sh: 1: xdg-user-dir: not found
```

It still works as expected, but might confuse users.

qhaas, 2019-07-24:
Seeing the same in 0.9.5 (Type 5 in my post), appears harmless, but yes, I don't like errors and it scared me at first too:  https://github.com/carla-simulator/carla/issues/1826

You can install a package of the name to clear it, but I'm not sure what the consequences of that are.

closed 2019-09-29 by stale[bot]

jrayner, 2020-10-11:
Hmm. This issue does appear to exist, at least for me.

$ docker run -p 2000-2002:2000-2002 --runtime=nvidia --gpus all carlasim/carla:0.9.10
sh: 1: xdg-user-dir: not found

docker ps doesn't show the server

With -it:

4.24.3-0+++UE4+Release-4.24 518 0
Disabling core dumps.
sh: 1: xdg-user-dir: not found

I don't know if this is nvidia dependent (nvidia/cuda 11.0-base image)

kargarisaac, 2020-10-14:
> Seeing the same in 0.9.5 (Type 5 in my post), appears harmless, but yes, I don't like errors and it scared me at first too: #1826
> 
> You can install a package of a similar name to clear it, but I'm not sure what the consequences of that are.

It exits after this error and goes to exited mode. I cannot see it using docker ps.

hellojql, 2020-10-23:
> Hmm. This issue does appear to exist, at least for me.
> 
> $ docker run -p 2000-2002:2000-2002 --runtime=nvidia --gpus all carlasim/carla:0.9.10
> sh: 1: xdg-user-dir: not found
> 
> docker ps doesn't show the server
> 
> With -it:
> 
> 4.24.3-0+++UE4+Release-4.24 518 0
> Disabling core dumps.
> sh: 1: xdg-user-dir: not found
> 
> Same happens without specifying the nvidia runtime:
> 
> $ docker run -p 2000-2002:2000-2002 carlasim/carla:0.9.10
> sh: 1: xdg-user-dir: not found

it happens on me as well

qhaas, 2020-10-27:
> It exits after this error and goes to exited mode. I cannot see it using docker ps.

Given newer versions of CARLA are compiled in 'Shipping' mode, the verbosity is reduced.  It is likely this warning is a 'red herring' since it is the last of the 'expected' output to the console.  Deploying the `xdg-user-dirs xdg-utils` packages inside the container should clear the warning and verify if it is causing the problem or not, here is a snippet on how to build a Docker image based on CARLA 0.9.8 (newer versions should work, just update the version in the `FROM` line), with said packages included.

```
$ cat Dockerfile 
FROM carlasim/carla:0.9.8
USER root
RUN apt-get update && apt-get install -y xdg-user-dirs xdg-utils && apt-get clean
USER carla
$ docker build -t test:carla-xdg .
...
$ docker run --rm -d --name test-carla-xdg-server --gpus=all test:carla-xdg
...
$ docker logs test-carla-xdg-server | wc -l
0
$ docker ps | grep carla
6011f2d916b4 test:carla-xdg "/bin/sh -c '/bin/ba…" 2 minutes ago Up 2 minutes test-carla-xdg-server
```

Disclaimer:  I'm not affiliated with CARLA, I'm just a user

yshen47, 2020-11-08:
I found a workaround finally to this issue. If you wrap the docker command with this: 
`SDL_VIDEODRIVER=offscreen SDL_HINT_CUDA_DEVICE=0 ./CarlaUE4.sh -ResX=800 -ResY=600 -nosound -windowed -opengl`
Then it is able to run the server. check #225

kannidekan, 2020-11-11:
> I found a workaround finally to this issue. If you wrap the docker command with this:
> `SDL_VIDEODRIVER=offscreen SDL_HINT_CUDA_DEVICE=0 ./CarlaUE4.sh -ResX=800 -ResY=600 -nosound -windowed -opengl`
> Then it is able to run the server. check #225

Thanks, it works for me!
[image]
[image]
And I can run Python scripts on another machine, but it is very slow, has it happened to you?

dhiegomaga, 2021-05-10:
@yshen47 
> I found a workaround finally to this issue. If you wrap the docker command with this:
> `SDL_VIDEODRIVER=offscreen SDL_HINT_CUDA_DEVICE=0 ./CarlaUE4.sh -ResX=800 -ResY=600 -nosound -windowed -opengl`
> Then it is able to run the server. check #225

What does "wrap the docker command" mean? This? 

`sudo nvidia-docker run -it --rm  carlasim/carla:0.9.10 SDL_VIDEODRIVER=offscreen SDL_HINT_CUDA_DEVICE=0 ./CarlaUE4.sh -ResX=800 -ResY=600 -nosound -windowed -opengl`

I get this error:

`docker: Error response from daemon: OCI runtime create failed: container_linux.go:367: starting container process caused: exec: "SDL_VIDEODRIVER=offscreen": executable file not found in $PATH: unknown.`

rzeydelis, 2022-01-15:
`xhost local:root` fixed this issue for me.

^^ Type this command first and then try running Carla again:
```
sudo docker run   -e SDL_VIDEODRIVER=x11   -e DISPLAY=$DISPLAY  -v /tmp/.X11-unix:/tmp/.X11-unix   -p 2000-2002:2000-2002   -it   --gpus all   carlasim/carla:0.9.10 ./CarlaUE4.sh -opengl
```

happytianhao, 2024-08-17:
apt-get install xdg-user-dirs