carla-simulator/carla #1896: In Carla you can define activities for pedestrians such as running, jumping, talking?

hvelesaca, 2019-07-16:
**In Carla you can define activities for pedestrians such as running, jumping, talking ?**

bernatx, 2019-07-17:
Hi,

Not right now in this version 0.9.6 but we are planning to add those features soon in future versions.

hvelesaca, 2019-07-17:
But in this version 0.9.6 you can Controlling a walker's skeleton and I will simulated the activities such as running, jumping, talking an others?

bernatx, 2019-07-18:
Yes, but that is more oriented to let the pedestrian animate for an action like talk, call a taxi, sit down...other than navigate or move around. For that, we need to take care of the collisions with the world, something that with the animation you can not do. But you are free to test :)

closed 2019-09-23 by stale[bot]