carla-simulator/carla #1899: [0.9.6] Make pedestrians cross streets

dianchen96, 2019-07-16:
Hi Carla team!

Is there any way to make the pedestrians in 0.9.6 cross the roads similar to 0.8? Right now they only them walk on sidewalks with the default pedestrian controller.

Thanks!

bernatx, 2019-07-17:
Hi,

That behavior will be added in future versions of the simulator, but right now the pedestrians are limited to walk on authorized zones only, mainly sidewalks.

dianchen96, 2019-07-18:
Thanks for your reply @bernatx! Is editing the navmesh file a temporary workaround for the current Carla version? Many thanks!

bernatx, 2019-07-19:
Currently we use a navmesh that has only sidewalks, that is why pedestrians can not walk out of them. All pedestrian navigation is done from the client, we use the Recast&Detour library (https://github.com/recastnavigation/recastnavigation) with which you can export a .BIN file with the demo tool they have. Then just replace that .BIN file in the Maps/Nav/ folder, with the name of the map you want to use. The .BIN is loaded by the server and sent to the client, who then use it to make all calculations for the pedestrians navigation. Currently they walk at random places without any logic, so I'm not sure if that will help you on your goal.

dianchen96, 2019-07-19:
Thanks so much @bernatx ! I exported the maps using the 'Export All' in the UE4 editor and saved them as .obj files. However when I opened them in RecastDemo they show up as a sphere instead of the correct meshes. Do I need to do anything before exporting? Sorry to bug again, thanks in advance!

[image]

bernatx, 2019-07-19:
Hi, some tips:

- Try to export only the geometry that you need, and hide all other. Then, with the geometry to export selected use the option 'export selected'. That way you will not select the skydome and other big geometry that will be a problem on Recast. 

- Also, take care of the scale, I had to adjust the code of Recast just to adjust the scale at 0.1f at the time of import the .OBJ because Unreal use centimeters while Recast use meters  (in rcMeshLoaderObj() constructor, set m_scale(0.1f)). Once the .BIN is exported it is in meters and does not need more scale adjustment, because it will be read only by Recast library, using meters.

I hope that help.

dianchen96, 2019-07-25:
Thanks so much @bernatx ! I followed your tips and labeled the intersections and part of other roads as walkable, and now the pedestrians can walk on roads! However the pedestrians don't seem to be avoiding the cars correctly, as shown in the first few seconds in this clip: https://drive.google.com/file/d/1Si3oZNrA1uhwIk5tzNaZ_6QjfOkf8czF/view?usp=sharing. 

Could you also kindly give some pointers on how to fix this? Should we look at the obstacle avoidance settings in the dtCrowd object? Many thanks!

bernatx, 2019-07-29:
Hi,
The pedestrians can not detect or avoid vehicles (or dynamic objects) yet because they are not included in the navigation calculations. We are planning to develop that feature and others in a future release. I think there is no simple workaround for that.

Siqi-Wang47, 2019-08-09:
hi @dianchen96, I am the new to Carla. Could you tell me how you make the pedestrians go cross the roads in detail? I will really appreciate for your time and effort.

dianchen96, 2019-08-10:
I have a custom carla version now where the pedestrians cross the roads and avoid the cars. I will create a fork and push in few days, in case anyone else also need this feature before the official fix.

Siqi-Wang47, 2019-08-12:
@dianchen96 That's so great! When you make it, please let me know because I'm stuck at this step. Thank you very much!

Siqi-Wang47, 2019-08-14:
Hi, @dianchen96, have you created a fork to push your custom carla version in? I am really in a hurry to finish it. If it's not available for you to do that, could you please tell me how you installed the Recast&Detour library and use it? I couldn't make it when following the guide on Github. 
Thank you very much!

ndeshp2s, 2019-08-15:
Hi @dianchen96, 
I am new to Carla and wanted to have exactly the same functionality (pedestrians crossing the road). So eagerly looking forward to your version of Carla where this functionality is available.

Also, @bernatx, I was considering to use the older version of Carla until the pedestrian crossing functionality is not added in newest version. So could you please tell me the last version where this functionality was available? 

Thank you.

dianchen96, 2019-08-19:
Here you go: https://github.com/dianchen96/carla/tree/dev-peds

- This is only a hack where vehicles are considered round circles. It can create problems if traffic is too dense, and the motion looks a bit unnatrual: https://drive.google.com/file/d/1hgnpi6olrYeevXLNKGB7g8l1YeldD6KU/view?usp=sharing

- For the peds to walk on streets you still need to get your own navmeshes using the recast&detour demo tool, following @bernatx 's guide. I created a custom one for Town01 [here](https://drive.google.com/file/d/1YjP8KAQp3kR0oeJpkVUXHahO2j8itm2b/view?usp=sharing) where peds cross the streets.  
- The dtcrowd object can fail to navigate for the peds if there are too many of them (or too many cars)
- I used a fork version of recast&detour because for complicated meshs it seems the dtcrowd need a larger buffer for computing the corridors. [This](https://github.com/recastnavigation/recastnavigation/blob/master/DetourCrowd/Source/DetourCrowd.cpp#L422) might need more tweaking if the navmesh is very complicated.

ndeshp2s, 2019-08-20:
Hi @dianchen96,

Thankyou so much for sharing the details. Even I have been  trying to generate the navmesh file through recast&detour. However, I am unable to build its demo version. Did you also face any difficulties while building it?

dianchen96, 2019-08-20:
@ndeshp2s @Siqi-Wang47 make sure you use gcc/g++ 8 or higher to compile.

ndeshp2s, 2019-08-21:
Thank you @dianchen96. This solved the issue.

Sorry to bother you. But I still have some doubts. 
1. I cannot figure out how to open the .obj file (I created through Unreal Engine). There is no "Tab Bar and Menu Bar" in  the Recast&Detour library. Am I doing something wrong here?
2. And once I  have imported the .obj file, how to I label the regions of my interest as walkable? I do not see any option for that either.

dianchen96, 2019-08-22:
Could you be more specific about the segfault? To compute the navmesh you can just hit the 'Build' button on the right of the demo window.

ndeshp2s, 2019-08-22:
Actually there is no information printed about the error. It just says segmentation fault. 
Did you also remove the sky sphere geometry object from unreal?

ndeshp2s, 2019-08-23:
Hi @dianchen96 ,

I managed to remove the sphere from recast, however, now I have another issue. After I hit the 'Build' 
button nothing happens. The recast library just freezes. And it does not create the required .bin file.

After opening the .obj file in recast I straightway hit the 'build' button with all the parameters as shown on the right side of the figures below. Am I missing out some thing here?

[image]

[image]

dianchen96, 2019-08-23:
I think you need to change the scale of the recast demo, following @bernatx 's tips above, and recompile the recast demo.

ndeshp2s, 2019-08-26:
Hi @dianchen96,
I tried changing the scale as well. Still no success.

ndeshp2s, 2019-08-28:
Hi @dianchen96 ,

I did try these options, however, it still does not work. I guess for now I will use your .obj file (for Town01) since I have already spend a lot of time on this and I need to have a basic simulation working soon.

But thank you so much for your help. I will update you if I get any success with this task.

ndeshp2s, 2019-09-02:
Hi @dianchen96,

Sorry to bother you again. You had  mentioned "The vehicles need to call vehicle.start_dtcrowd() to add themselves to the dtcrowd." I cannot figure exactly where this method needs to be called.

Could you please guide about the same? 

Thank you.

closed 2019-11-08 by stale[bot]

Mkarami3, 2020-06-15:
Dear ndeshp2s,

Did you solve the problem in "RecastDemo". When I push "Build" button, it does not create the bin file.
Thanks

MasterTigress, 2021-04-11:
Hi all, I am wanting to make the Pedestrian spawned and walk directly towards the vehicle for my research. I wrote up a scratch up code that would spawn the Pedestrian right in the middle of the road and should walk towards the vehicle. The Pedestrian is spawned in the middle of the road, however when the Pedestrian walks it suddenly shifts (kind of reappear) to the left side of the road and is walking towards the sidewalk on the right side. From this thread I understand that there is a navigation restriction for areas that Pedestrian can walk. Is it possible to completely disable this restriction in an easy way so that my Pedestrian can be spawned at the middle of the road and walk towards the vehicle ? My intention is to capture human walk radar data.
Please recommend a solution

badr-ben, 2021-11-06:
Hi all, Please I want to create my own scenario in Carla 0.9.12.  Do I use the Recast&Detour library or not?

sbv198, 2022-06-13:
@MasterTigress  Did you manage to do the coding that you were trying to do. If yes ciould you share little bit more into details, i am also working on similar tjing

shahbazsiddique69, 2024-12-03:
hi @ndeshp2s 
Have you solved the build button error?
My recast also freezes when i click the build button and it doesn't create any binary file as well.
thank you