carla-simulator/carla #1905: RuntimeError: rpc::rpc_error during call in function version

asharalam11, 2019-07-17:
I have build Carla from source on Ubuntu 16.04. 
I can open the maps in the Carla Unreal Editor but I get an error whenever I try to run a Python file from Python API/examples/

Traceback (most recent call last):
  File "./manual_control.py", line 870, in <module>
    main()
  File "./manual_control.py", line 862, in main
    game_loop(args)
  File "./manual_control.py", line 783, in game_loop
    world = World(client.get_world(), hud, args.filter, args.rolename)
RuntimeError: rpc::rpc_error during call in function version

PS: I also had a Carla release version working with Python before but since I installed the source one, even the earlier release version shows the same behavior with Python API examples.

Can anybody help????

asharalam11, 2019-07-17:
Figured it out! I should have used python3 instead of just python

closed 2019-07-17 by asharalam11

Kailiangdong, 2019-08-26:
> Figured it out! I should have used python3 instead of just python

What version is your python?man. I still have this problem in Ubuntu 16.04 for carla 0.9.6

iamycee, 2019-09-02:
I am using Python 3.6.x on Ubuntu 18.04 with CARLA 0.9.6 and getting the same error when I run `manual_control.py` or `tutorial.py`

Kailiangdong, 2019-09-02:
@iamycee try to use python 3.5

karthi0804, 2019-09-27:
I built from source and using python 3.5 and carla 0.9.6 version. Still facing this issue. Can someone help me?

Kailiangdong, 2019-09-29:
@karthi0804 What's you system? I later found my mistake is from that I wrongly "pip install carla" in 0.9.6. It should use egg to install package of carla

MalathiKaliyappan, 2020-08-25:
Hi @Kailiangdong , I am using python 3.6 on Ubuntu 18.04 with CARLA 0.9.5.
I got the same error while running spawn_npc.py. 
```
 python3 spawn_npc.py -n 80

destroying 0 actors

done.
Traceback (most recent call last):
  File "spawn_npc.py", line 124, in <module>
    main()
  File "spawn_npc.py", line 70, in main
    world = client.get_world()
RuntimeError: rpc::rpc_error during call in function version
```

Can you please help me to solve this?

navjotkaurdhoat, 2021-07-20:
> Hi @Kailiangdong , I am using python 3.6 on Ubuntu 18.04 with CARLA 0.9.5.
> I got the same error while running spawn_npc.py.
>
> ```
>  python3 spawn_npc.py -n 80
>
> destroying 0 actors
>
> done.
> Traceback [...]

I am getting the same error. Were you able to resolve it? Or anyone else who can help. Any help would be appreciated.

Thanks!

6Lackiu, 2022-04-22:
Just tested solution!
I start manual_control.py encountered the same problem. I found that the egg file was not added to the environment variable. At first, it was like this: `export pythonpath = "${carla_root} / Python API / Carla /:" ${scenario_runner_root} ":" ${leaderboard_root} "`
Changed to: ` export pythonpath = "${carla_root} / Python API / Carla /:" ${scenario_runner_root} ":" ${leaderboard_root} ":" ${carla_root} / Python API / Carla / dist / carla-0.9.10-py3.7-linux-x86_64. Egg ": ${pythonpath}`
Finally, you should start manual_control.py through `python3` instead of `python`! This perfectly solves the problem, at least for me.
My environment is Ubuntu 16 04, Carla version is 0.9.10.1. I started Carla from a remote server, so I added `-opengl`.
I hope everyone can solve the problem!

MuHeDing, 2024-10-09:
> Just tested solution! I start manual_control.py encountered the same problem. I found that the egg file was not added to the environment variable. At first, it was like this: `export PYTHONPATH= " [...]

good suggestions!