carla-simulator/carla #1908: How to change bundled map

atinfinity, 2019-07-18:
## System information (version)
- Ubuntu 18.04 64bit
- CARLA v0.9.6(compiled version)

## Detailed description
I tried to change bundled map using CARLA v0.9.6.  
Because, this specification is not reflected.  

By the way, it was able to reflect in the previous version(v0.9.5).  
Has the option to specify bundled map changed?

## Steps to reproduce
### Simulator
```shell
$ cd CARLA_0.9.6
$ ./CarlaUE4.sh Town01 -windowed -ResX=320 -ResY=240 -benchmark -fps=10
```

### Python script
```bash
$ cd CARLA_0.9.6/PythonAPI/examples
$ python manual_control.py
```

I tried to use `Town01` via option of simulator.  But, `Town03` is selected.

togaen, 2019-07-20:
I am also having this problem with 0.9.6. Previously, in 0.9.5, I specified the desired map at launch time:

```
./CarlaUE4.sh /Game/Carla/Maps/Town06 -windowed -ResX=160 -ResY=120 -quality-level=Low
```

But that no longer works. The documentation says the python API in 0.9.6 can be used to load a map:

```
world = client.load_map('Town01')
```

but when I try that method, I get: `TypeError: 'NoneType' object is not iterable`

togaen, 2019-07-20:
At least for the python API, it seems like the documentation was just wrong. The command for loading a map should be `load_world`. Pull request here: #1914

atinfinity, 2019-07-21:
@togaen Thank you for your help! I changed [PythonAPI/examples/manual_control.py](https://github.com/carla-simulator/carla/blob/0.9.6/PythonAPI/examples/manual_control.py) to change map.

```diff
 def game_loop(args):
     pygame.init()
     pygame.font.init()
    world = None

     try:
         client = carla.Client(args.host, args.port)
         client.set_timeout(2.0)
+        client.load_world('Town01')
+        client.reload_world()
```

As a result, I was able to change the map to `Town01`.  
It seems that I need to call `reload_world()`.

togaen, 2019-07-23:
According to @jackbart94, the map can no longer be set at launch time. Instead, he says:

You now have to launch

`./CarlaUE4.sh`

then on another terminal go to PythonAPI/util and launch 

`python config.py -h`

and it will give you the list of everything you can change, but now at runtime. For example, to put Town05 do 

`python config.py -m Town05`

bernatx, 2019-07-23:
Yes, the problem is that now in Linux the package is now 'shipping' mode instead of 'development' mode like previously. The first is better in performance, but at the cost that is difficult for debugging and it does not allow using command line parameters when launching ./CarlaUE4.sh. 

So, all the command line parameters have been moved to the config.py script, which also includes a few more parameters. I think we don't have all these well explained, so we will check the documentation.

Thanks for telling.

atinfinity, 2019-07-24:
@bernatx @togaen Thank you for your comment. I'll trye to use `config.py`.

closed 2019-07-24 by atinfinity

togaen, 2019-07-24:
@bernatx Thanks for the explanation. Yeah, it would be great to update the docs and make it clearer. Also, it would be really helpful if the launch script emits a warning or something if someone passes in an argument that is no longer supported. When I initially noticed the map argument wasn't working, I thought my installation was corrupt and I spent a lot of time verifying that it was not.