carla-simulator/carla #1911: import assets/maps in Carla 0.9.6

zhang-xinxin, 2019-07-19:
I have some customized maps created by RoadRunner. I've imported them into Carla 0.9.4 and Carla 0.9.5 successfully using the previous import pipleline. 

Now the import process has been further simplified in Carla 0.9.6. I want to import my customized maps into Carla 0.9.6 using the new import pipeline, but failed.

According to your guideline, I put my map under the Import folder, created the .json file,  and then "make import". The printed messages seems all is ok. Then I launch the Unreal editor and load the new map, which looks like below snapshot. It seems like all textures are lost.   

[image]

If I play the map, no road/mesh display, just like the below snapshot. 
[image]

Could you please give me some advice on that? Thanks very much!

bernatx, 2019-07-19:
Probably the scale of the FBX needs to be 10x greater (from meters to cm). The info about the open drive seems correct, right?

zhang-xinxin, 2019-07-22:
Hi @bernatx, thanks very much for your attention and reply. The OpenDrive info is correct, but no texture display. 

In order to address the problem, I did some experiments.

test1: 
        create a new level in UE4 editor, and import the fbx file manually, and then select all assets and drag them to the new level.  The map is generated successfully.

test2:
        use the new process pipleline in Carla 0.9.6.  I prepared the json file and map folder firstly, and 'make import', then launched the UE4 editor and opened the imported map. Just as the snapshot in my last post, no texture displayed. But I found all assets were saved in the same folder with the .umap file. So I selected all assets and drag them to the imported level. Now I can see the texture (including trees, traffic light etc.), but the texture is 180-degree rotated comparing to the road mesh. 
[image]

So my questions are : 
    1. why the imported assets weren't added into the created level automatically?
     2. How to adjust the scale and rotation of the fbx?

Here is my json file:

{
    "maps": [{
        "name": "case1_map",
        "source": "./case1_map/case1_map.fbx",
        "use_carla_materials": true,
        "xodr": "./case1_map/case1_map.xodr"
      },
      {
        "name": "case2_map",
        "source": "./case2_map/case2_map.fbx",
        "use_carla_materials": true,
        "xodr": "./case2_map/case2_map.xodr"
      }],
    "props": []
}

Thanks!

manishthani, 2019-07-22:
Hi @zhang-xinxin ,

Regarding the issue related with the map that disappears when going in Play Mode in Unreal Engine, maybe it could be related with the `BaseMap` located inside the `Content/Carla/Maps/BaseMap` folder. 

Basically, this Map is used to cook assets and maps, and maybe it could happen that it has been altered during a importing process done by the `PrepareAssetsForCooking` commandlet. 

Please, could you check if it has been modified and replace it by the one located in the latest version of the Carla Content?

zhang-xinxin, 2019-07-23:
Hi @manishthani , thanks very much for your reply. 

The issue isn't just in the Play Mode.  Due to some unknown reasons, the map importing process (https://carla.readthedocs.io/en/latest/asset_packages_for_dist/) doesn't work in my case. After 'make import' completed, only waypoints displayed, but no assets/textures showed in the map.

I checked the BaseMap folder, all files remained unmodified. 
If nessary, I can attach my map files here.

manishthani, 2019-08-01:
Hi @zhang-xinxin ,
Sorry for late reply. Yes, I would really appreciate if you could attach those files here so I can have a look at it and see what happens! Thank you very much! :)

zhang-xinxin, 2019-08-13:
Hi @manishthani, sorry for late reply, I just saw your message today.  
I've uploaded my map folder to google drive and shared it with you.  Please kindly try the following web link:

https://drive.google.com/file/d/1mIHNmBc6BpxeqvbubS63gqhVDPZTUZkD/view?usp=sharing

sheelabhadra, 2019-08-20:
Hi @zhang-xinxin @manishthani, I'm facing the exact same issue. Were you able to find out a way to solve it?

manishthani, 2019-08-21:
Hi everyone,

Just fixed the issue regarding the scale of the map, so that it appears in the correct size automatically when importing the map. There was missing an extra flag in the importing script that converted the units from FBX units to UE4 units (i.e. from meters to centimeters, respectively). I tested it with @zhang-xinxin map and in Play Mode, and now it is possible to visualize the map. You can see the changes in branch `manishthani/import_assets`. Please, let me know if now it works for you as well.

Regarding the issue of adding assets in a new level automatically, technically, the importing process should add the assets of the Marking Nodes, Road Nodes and Terrain Nodes automatically from the map exported by RoadRunner. However, in your map that is not happening, so I will investigate more on that particular case and keep you updated.

Thanks a lot!

sheelabhadra, 2019-08-21:
Hi @manishthani, thanks a lot for working toward resolving the issue. I'm using your code and can add the assets manually on Unreal Editor. But, it's a tedious process. Please keep me posted on your progress toward solving the issue of importing assets automatically. Thanks again!

Lunaticsoulsaran, 2019-08-26:
Hello ,
Step 1: 
I have created a customized map using Open streetmap ,then i have converted the .osm file to .xodr using sumo converter.
Step 2:
Then i have exported the corresponding .fbx file through blender.
Step 3:
As per the carla documentation i have imported the fbx file manually, and then select all assets and drag them to the new level. The map is generated successfully.
[image]
Step 4 : 
When play it ,black screen is displayed.
[image]
Could someone help me out,Thank a lot in advance.

manishthani, 2019-08-29:
Hi everyone,

I fixed the issue regarding adding assets automatically in branch `manishthani/import_assets`. Please test it and let me know if it worked.

However, that led to another issue: When trying to move the imported assets to the corresponding folder for semantic segmentation, the assets loose the references to its materials, so when they are spawned, no material is found. Apparently, it is something related with commandlets in Unreal Engine and already posted a question in their forum to get a solution to it, so this is something I will be working on the following days. If you need semantic segmentation urgently as well, what you could do meanwhile as a workaround is to separate them in folders manually using the Editor once these assets are imported, that worked for me when testing it.

Thank you very much and will keep you updated!

zhang-xinxin, 2019-08-30:
Hi @manishthani Thanks for your great effort! I've test it and it works for me. And looking forward to your further message.

eloy96vg, 2019-09-02:
Thanks @manishthani , this also solved my issue ( #1886 ). Now, i can import maps from RoadRunner without problems.

manishthani, 2019-09-03:
Hi everyone,

Happy to hear that! You are welcome! :)

I came up with a fix for the bug of moving the assets of a map automatically in the corresponding semantic segmentation folder without loosing references to their corresponding material in branch `manishthani/import_assets`. So basically, if an asset's name belonging to a map contains the words `RoadNode`, `Terrain` or `MarkingNode` (These names are part of the nomenclature specified by RoadRunner when it exports a map as `.FBX`), the import process will move the meshes to the `Roads`, `Terrain` and `Roadlines` folder inside `Content/PackageName/Static`, respectively, for semantic segmentation. If there is no match, it will place the assets inside `Content/PackageName/Maps/MapName`. If not satisfied with this automatic process, you can always move the assets to the folders you wish afterwards.

  I would really appreciate if you could test it and provide me your feedback. We will update the documentation about this once it is fully tested.

Thank you very much for your support!

closed 2019-09-17 by manishthani

Kingsman1995, 2019-10-07:
hello, may i ask where did u get the RoadRunner install package? i tried to contact vectorzero, but no response yet.

sourabhmisal, 2020-07-02:
In the [video](https://www.youtube.com/watch?time_continue=6&v=dvPhN0Gbb0w&feature=emb_logo), I see that there is a branch of bernet/fix_import, which is used to run the docker file, is that branch merged or still pending?

ronyshaji, 2022-11-18:
> Probably the scale of the FBX needs to be 10x greater (from meters to cm). The info about the open drive seems correct, right?

@bernatx Hey, just a doubt while going through this. How can we import a custom .uasset blueprint file build in unreal engine to carla ? i just drag and droped it into the level and it is working but it is not detecting so that I can customise with python API:. Is there any other way to import .uasset file to carla?

AndritMuca, 2023-07-02:
@manishthani Please could you have a look at these issue: https://github.com/carla-simulator/carla/issues/6597
Its almost similar to the issue you have solved, but i did not use Roadrunner for export fbx and xodr. Thank you very much!

PR #2068 (2019-09-17): Fixed bugs of importing assets

#### Description
Refactored code and fixed issues regarding scale, materials and semantic segmentation when importing maps to Unreal Engine.

Fixes #1911 and  #1886 

#### Where has this been tested?

  * **Platform(s):** Ubuntu 16.04LTS
  * **Python version(s):** 2.7.12 and 3.5.2
  * **Unreal Engine version(s):** 4.22
#### Possible Drawbacks
Import process might take more time since the `import.py` script will invoke the `MoveAssetsCommandlet` for moving the meshes, and then `PrepareAssetsForCookingCommandlet` for preparing the map. This is done since moving me [... cut]