carla-simulator/carla #1917: Run CARLA in RED HAT.

arvindpalkyutech, 2019-07-22:
I want to run CARLA in Red Hat. I have installed docker in Red Hat and also I have Ubuntu image in installed docker, but installing Docker-CE in ubuntu image shows many package missing. Can anyone please guide me for this?

qhaas, 2019-07-24:
I'm able to run CARLA 0.9.5 in Docker in CentOS 7 (RHEL7 clone).  If you installed / updated Docker / nvidia-docker in the last few days, be aware of some ongoing issues with the recent nvidia-docker release:
https://github.com/NVIDIA/nvidia-docker/issues/1017
https://github.com/NVIDIA/nvidia-container-runtime/issues/68

Here is my environment
```
$ cat /etc/redhat-release 
CentOS Linux release 7.6.1810 (Core)

$ rpm -qa '*docker-ce*'
docker-ce-19.03.0-3.el7.x86_64
docker-ce-cli-19.03.0-3.el7.x86_64

$ rpm -qa '*nvidia*'
libnvidia-container-tools-1.0.2-1.x86_64
nvidia-x11-drv-430.34-1.el7_6.elrepo.x86_64
yum-plugin-nvidia-1.0.2-1.el7.elrepo.noarch
nvidia-container-toolkit-1.0.1-2.x86_64
kmod-nvidia-430.34-1.el7_6.elrepo.x86_64
libnvidia-container1-1.0.2-1.x86_64
nvidia-x11-drv-libs-430.34-1.el7_6.elrepo.x86_64
```
Here is what I ran, note that this will likely change with CARLA 0.9.6 and I had to switch `--runtime=nvidia` to `--gpus all` per the recent nvidia-docker changes:  
`docker run -p 2000-2002:2000-2002 --gpus all carlasim/carla:0.9.5 /bin/bash CarlaUE4.sh /Game/Carla/Maps/Town03 -benchmark -carla-server -fps=10 -world-port=2000 -windowed -ResX=100 -ResY=100 -carla-no-hud`

arvindpalkyutech, 2019-08-02:
> I'm able to run CARLA 0.9.5 in Docker in CentOS 7 (RHEL7 clone). If you installed / updated Docker / nvidia-docker in the last few days, be aware of some ongoing issues with the recent nvidia-docker release:
> [NVIDIA/nvidia-docker#1017](https://github.com/NVIDIA/nvidia-docker/issues/1017)
> [NVIDIA/nvidia-container-runtime#68](https://github.com/NVIDIA/nvidia-container-runtime/issues/68)
> 
> Here is my environment
> 
> ```
> $ cat /etc/redhat-release 
> CentOS Linux release 7.6.1810 (Core)
> 
> $ rpm -qa '*docker-ce*'
> docker-ce-19.03.0-3.el7.x86_64
> docker-ce-cli-19.03.0-3.el7.x86_64
> 
> $ rpm -qa '*nvidia*'
> libnvidia-container-tools-1.0.2-1.x86_64
> nvidia-x11-drv-430.34-1.el7_6.elrepo.x86_64
> yum-plugin-nvidia-1.0.2-1.el7.elrepo.noarch
> nvidia-container-toolkit-1.0.1-2.x86_64
> kmod-nvidia-430.34-1.el7_6.elrepo.x86_64
> libnvidia-container1-1.0.2-1.x86_64
> nvidia-x11-drv-libs-430.34-1.el7_6.elrepo.x86_64
> ```
> 
> Here is what I ran, note that this will likely change with CARLA 0.9.6 and I had to switch `--runtime=nvidia` to `--gpus all` per the recent nvidia-docker changes:
> `docker run -p 2000-2002:2000-2002 --gpus all carlasim/carla:0.9.5 /bin/bash CarlaUE4.sh /Game/Carla/Maps/Town03 -benchmark -carla-server -fps=10 -world-port=2000 -windowed -ResX=100 -ResY=100 -carla-no-hud`
> 
> I too am getting many warnings / errors, but they appear to be harmless: #1826

Thank you qhaas it worked for me...thanks a lot

closed 2019-10-08 by stale[bot]