carla-simulator/carla #1919: 'PythonAPI' failed

OmarElzaafaranyBrightskies, 2019-07-22:
Hello Everyone,
Unfortunately, my "make PythonAPI" command fails.

Carla version 0.9.6
Ubuntu 18.041
Unreal Engine 4.22.3

The error message is below. Can you help me?

make PythonAPI  # Compiles the PythonAPI module necessary for running the Python examples.
Setup.sh: llvm-7.0 already installed.
Setup.sh: boost-1.69.0-c7 already installed.
Setup.sh: rpclib-v2.2.1_c2-c7 already installed.
Setup.sh: gtest-1.8.1-c7 already installed.
Setup.sh: recast-c7 already installed.
Setup.sh: CARLA version 0.9.6-dirty.
Setup.sh: Generating CMake configuration files.
Setup.sh: Success!
BuildLibCarla.sh: Building LibCarla "Client.Release" configuration.
ninja: no work to do.
[0/1] Install the project...
-- Install configuration: "Client"
BuildLibCarla.sh: Success!
BuildPythonAPI.sh: Building Python API for Python 2.
compiling:
  - source/libcarla/libcarla.cpp
running bdist_egg
running egg_info
writing source/carla.egg-info/PKG-INFO
writing top-level names to source/carla.egg-info/top_level.txt
writing dependency_links to source/carla.egg-info/dependency_links.txt
reading manifest file 'source/carla.egg-info/SOURCES.txt'
writing manifest file 'source/carla.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-x86_64/egg
running install_lib
running build_py
running build_ext
building 'carla.libcarla' extension
clang-6.0 -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -Wdate-time -D_FORTIFY_SOURCE=2 -g -fdebug-prefix-map=/build/python2.7-MW0004/python2.7-2.7.15=. -fstack-protector-strong -Wformat -Werror=format-security -fPIC -Idependencies/include -I/usr/lib/gcc/x86_64-redhat-linux/7/include -I/usr/include/python2.7 -c source/libcarla/libcarla.cpp -o build/temp.linux-x86_64-2.7/source/libcarla/libcarla.o -isystem dependencies/include/system -fPIC -std=c++14 -Werror -Wall -Wextra -Wpedantic -Wno-self-assign-overloaded -Wdeprecated -Wno-shadow -Wuninitialized -Wunreachable-code -Wpessimizing-move -Wold-style-cast -Wnull-dereference -Wduplicate-enum -Wnon-virtual-dtor -Wheader-hygiene -Wconversion -Wfloat-overflow-conversion -DBOOST_ERROR_CODE_HEADER_ONLY -DLIBCARLA_WITH_PYTHON_SUPPORT -DLIBCARLA_IMAGE_WITH_PNG_SUPPORT=true
error: unknown warning option '-Wno-self-assign-overloaded'; did you mean
      '-Wno-self-assign-field'? [-Werror,-Wunknown-warning-option]
error: command 'clang-6.0' failed with exit status 1
Util/BuildTools/Linux.mk:80: recipe for target 'PythonAPI' failed
make: *** [PythonAPI] Error 1

cassfalg, 2019-07-22:
Which version of clang do you have installed? Your error message mentions clang-6.0 but the Linux Instructions try to install clang-7.

OmarElzaafaranyBrightskies, 2019-07-22:
At first, I installed clang-7 then I installed clang-6 as it was requested after that.

WolfgangFahl, 2019-07-22:
Basically I have the same issue on Mac OS 10.13.6 using macports. Even after installing clang-6.0 with

`     sudo port install clang-6.0`

the problem persists.
```
unable to execute 'clang-6.0': No such file or directory
error: command 'clang-6.0' failed with exit status 1

```

analog-cbarber, 2019-07-22:
On Mac, I would recommend using the version of Xcode used to build the standard Unreal Engine distribution (e.g. Xcode 10.1 for Unreal 4.22) and then just use the system clang.

WolfgangFahl, 2019-07-22:
@analog-cbarber - thx for the hint but how do i tell the make system to do this?

analog-cbarber, 2019-07-22:
It is documented in Docs/how_to_build_on_mac.md on my fork.

You need to install the right version of Xcode, which you may have to get from developer.apple.com. If you have more than one version of xcode installed. You need to select the appropriate version using `sudo xcode-select`, which will set `/usr/bin/clang` to the appropriate version.

In any case, I would not expect you to be able to compile 0.9.6 from the master branch on a Mac. You will need a number of changes that are on my fork. I believe that my mac-build-0.9 branch is synced with CARLA 0.9.4. Is that what you are trying to build?

WolfgangFahl, 2019-07-22:
I'm going from https://github.com/carla-simulator/carla/issues/150 and have commented there that my progress can be seen at http://wiki.bitplan.com/index.php/Car_simulation_software#Carla. I am using the mentioned branch and xcode in principle. I could also use clang-7.0 or clang-6.0 from the macports project if that helps. The issues seem to be endless - is that the reason "EpicGames" used "epic" in their name? :-)

analog-cbarber, 2019-07-22:
I don't know what "going from #150" means given that there are a number of different suggestions for how to build. I have explicit instructions on my fork. Mixing and matching various peoples suggestions may not work. 

Note that if you change anything in your environment you may need to blow away your Build subdirectory and build from scratch.

OmarElzaafaranyBrightskies, 2019-09-22:
I have found that it was because of the network firewall. Although, it was working well with clang6 and clang8 servers. It was preventing the connection with Clang 7 servers.  When I changed the network the issue is solved.

closed 2019-09-22 by OmarElzaafaranyBrightskies