carla-simulator/carla #1920: make PythonAPI file using macports on MacOS

WolfgangFahl, 2019-07-22:
I am following the instructions at https://github.com/analog-garage/carla/blob/mac-build-0.9/Docs/how_to_build_on_mac.md and have documente my progress at 
http://wiki.bitplan.com/index.php/Car_simulation_software#Carla

The first part of the make process already runs successfully.

   make PythonAPI

fails with the error message:

carla-mac/Build/boost-1.69.0-install/include/boost/gil/extension/io/png/tags.hpp:25:14: fatal error: 
      'png.h' file not found
    #include <png.h>
             ^~~~~~~
1 error generated.

Since I installed libpng with macports the png.h file is at the standard macports location /opt/local/include:

```
cd /opt
find . -name png.h
./local/include/libpng16/png.h
./local/include/png.h

```

It is just not picked up by the build process.

I tried:

1. make PythonAPI  -I /opt/local/include
2. export CFLAGS=-I/opt/local/include;make PythonAPI
3. including via https://cmake.org/cmake/help/v3.0/command/include_directories.html in the top CMakeLists.txt 

but none of this works. The build system seems to expect a different way how to specify the include path for the PythonAPI target. Please advise.

WolfgangFahl, 2019-07-22:
For the time being I am trying a workaround with

```
cd /usr/local
sudo mv include include.org
ln -s /opt/local/include include

```

analog-cbarber, 2019-07-22:
I used brew to install libpng, which puts it in `/usr/local/lib`. Why don't you just use brew?

WolfgangFahl, 2019-07-22:
All my machines have macports installed. There are many use cases for macports and it works well. I never tried brew and wouldn't want to switch for this single use case.

analog-cbarber, 2019-07-22:
I understand, but using brew is a lot less evil than hacking your /usr/local/include directory!

If you want to add to the C include/lib paths you should modify Setup.command

analog-cbarber, 2019-07-23:
I would not expect this bug to ever get fixed. If you want to use macports, it is going to be up to you to modify the include/lib paths accordingly.

closed 2019-09-28 by stale[bot]