carla-simulator/carla #1921: Fix "Building Carla" section of README

WolfgangFahl, 2019-07-23:
As of 2019-07-23 the README section "Building Carla" has three statements on operating systems:

1. How to build on Linux 
2. How to build on Windows.
3. Unfortunately we don't have official instructions to build on Mac yet, please check the progress at issue #150

Given the long list of open issues with building on different environments I suggest to modify this section and list the options:

1. Linux
2. Windows
3. Mac OS
4. Docker

and add
- a necessary prerequisites
- a troubleshooting / FAQ section
- a description of what to expect 
for each

The motivation for this proposal is the frustration I got from trying to install carla on different MacOS computers to no avail.

analog-cbarber, 2019-07-23:
MacOS is not supported yet, and I do not believe the official maintainers use Macs for this project, so it doesn't seem reasonable to ask for instructions on how to build it at this point.

The best you can hope for is to duplicate builds from the few of us that have been able to get it to work  . If you stray too far from our instructions you are much more likely to encounter problems.

I have been working on upgrading my fork to 0.9.6 and will post my changes as soon as I get it mostly working, but I am not going to have time to try building in the wide variety of OS versions and tool configurations that other Mac users might want to use. There are some C++ fixes required to get it to work (mostly uint64/std::size_t issues).

closed 2019-09-28 by stale[bot]

mainakchain, 2020-10-13:
Has anything updated regarding running carla on Mac? Please let me know. I really want to make it run on my Mac. @WolfgangFahl @analog-cbarber

analog-cbarber, 2020-10-13:
I have not tried building on a mac for more than a year. It is definitely possible, but it is a bit of work. I don't have time to work on it right now or have a reason to use my regular work hours to work on that.

While it can be convenient to be able to do development on your mac, CARLA will really run much better on a machine with better GPUs, so I strongly recommend that you get a suitable linux or Windows box with the best GPU you can afford.