carla-simulator/carla #1927: Map of spawn points with indices in town03

elishafer, 2019-07-24:
Hi,

I need a map of the spawn points with indices for town03. Currently, using the no_rendering.py example I have the spawn points but I have no idea how to spawn at a certain point without an index. I've looked at other issues, but they were of no help.

Thanks!

ThrashAbaddon, 2019-08-23:
@elishafer 
Do you maybe mean waypoints?
My wild guess is that you want to see enumerated waypoints, and to get them enumerated you just need to, well, enumerate them.
```py
for index, wp in numerate(waypoints):
    # now draw your `wp` with text being the `index`
```
, if you have dynamic number of waypoints then create a global counter for them and just increment by 1 when you draw one waypoint.

elishafer, 2019-08-25:
Thanks, I actually meant was getting a map of spawn points for vehicles. In the example code for Python, actors are generated at spawn points. I'd like to know where these spawn points are located. Sort of like the example in the following picture which was taken from [documentation from the stable version of carla](https://carla.readthedocs.io/en/stable/benchmark_creating/):
[image]

marcgpuig, 2019-08-26:
Hi @elishafer.
Please, take a look at the function [`map.get_spawn_points`](https://carla.readthedocs.io/en/latest/python_api/#carla.Map.get_spawn_points), you can use it to get the spawn points regardless of the map.

Example:
```py
import random

# ...

client = carla.Client('localhost', 2000)
client.set_timeout(2.0)
world = client.get_world()
current_map = world.get_map()

# spawn_transforms will be a list of carla.Transform
spawn_transforms = current_map.get_spawn_points()
# get a single random spawn transformation over the map
random_spawn = random.choice(spawn_transforms)
```

Cheers!

cassfalg, 2019-08-28:
I think what's missing is the current version equivalent of `carla/planner/Town01.png` for each Town and of `view_start_positions.py` (unfortunately, 0.8.x Python API). I was actually looking for similar things. Yes, we can ask for the **list** of spawn points, but we can not **visualize** them because we do not have a rough map layout **picture** which tells us **where** in the map the coordinates of a spawn point are. How did you create those png maps? Can they be recreated?

elishafer, 2019-08-30:
@marcgpuig cassfalg articulated what I couldn't. I needed a way to visualise the spawn points. 
Thanks @cassfalg. Unfortunately the png that I posted is from an earlier version of Carla and I used that a long time ago so I don't remember.

BTW, this isn't so relevant for me now. I'm keeping this open in case someone out there also runs into this problem in the future.

cassfalg, 2019-09-02:
An Unreal Engine static minimap could be a way to create such overview images. There is documentation available online for it, though the use case is different and it would need to be adapted. I'm not that familiar with the Unreal Editor, so I may or may not investigate further. Dropping the hint here, maybe someone else wants to give it a try.

Maybe this is also a topic for the creative / map design team? At least the minimap picture?

cassfalg, 2019-11-04:
I would like to bump this, since I'd really like a minimap like image like we've discussed here.

cassfalg, 2020-03-03:
And again.

ThrashAbaddon, 2020-05-03:
.

germanros1987, 2020-06-26:
@cassfalg @ThrashAbaddon why not just using the functionality provided by the non_rendering_mode script? You can add the  '--show-spawn-points' attribute... 

If you want a cool top view, why don't you make a contribution to CARLA?

I consider the non_rendering_mode script to provide a solution to this problem, so I am closing this issue.

closed 2020-06-26 by germanros1987

PTorr, 2020-09-14:
I wrote a solution to all maps for this:
Got to PythonAPI --> examples --> no_rendering_mode.py
Change the file content with the code below.

Run with configuration: --show-spawn-points --map TownXX

Here is an example for Town05:
[image]

#!/usr/bin/env python

# Copyright (c) 2019 Computer Vision Center (CVC) at the Universitat Autonoma de
# Barcelona (UAB).
#
# This work is licensed under the terms of the MIT license.
# For a copy, see <https://opensource.org/licenses/MIT>.

# Allows visualising a 2D map generated by vehicles.

"""
Welcome to CARLA No-Rendering Mode Visualizer

    TAB          : toggle hero mode
    Mouse Wheel  : zoom in / zoom out
    Mouse Drag   : move map (map mode only)

    W            : throttle
    S            : brake
    AD           : steer
    Q            : toggle reverse
    Space        : hand-brake
    P            : toggle autopilot
    M            : toggle manual transmission
    ,/.          : gear up/down

    F1           : toggle HUD
    I            : toggle actor ids
    H/?          : toggle help
    ESC          : quit
"""

# ==============================================================================
# -- find carla module ---------------------------------------------------------
# ==============================================================================

import glob
import os
import sys

try:
    sys.path.append(glob.glob('../carla/dist/carla-*%d.%d-%s.egg' % (
        sys.version_info.major,
        sys.version_info.minor,
        'win-amd64' if os.name == 'nt' else 'linux-x86_64'))[0])
except IndexError:
    pass

# ==============================================================================
# -- imports -------------------------------------------------------------------
# ==============================================================================

import carla
from carla import TrafficLightState as tls

import argparse
import logging
import datetime
import weakref
import math
import random
[... 1559 lines cut ...]
        '--show-connections',
        action='store_true',
        help='show waypoint connections')
    argparser.add_argument(
        '--show-spawn-points',
        action='store_true',
        help='show recommended spawn points')

    # Parse arguments
    args = argparser.parse_args()
    args.description = argparser.description
    args.width, args.height = [int(x) for x in args.res.split('x')]

    # Print server information
    log_level = logging.DEBUG if args.debug else logging.INFO
    logging.basicConfig(format='%(levelname)s: %(message)s', level=log_level)

    logging.info('listening to server %s:%s', args.host, args.port)
    print(__doc__)

    # Run game loop
    game_loop(args)

if __name__ == '__main__':
    main()

JinghangLi, 2022-08-14:
If you want to plot the spawn points of your map, take a look at this [carla_dataset_tools](https://github.com/KevinLADLee/carla_dataset_tools). Run [visulize_map.py](https://github.com/KevinLADLee/carla_dataset_tools/blob/master/utils/visulize_map.py) and you will get a picture of respawn points like this
[image]
. If you like it, please star this repo, thanks.

blankdean, 2023-05-23:
Given the following code,
```
client = carla.Client('localhost', 2000)
client.set_timeout(10)
world = client.load_world('Town06')
current_map = world.get_map()

spawn_points = current_map.get_spawn_points()
```

The locations in spawn_points are out of my expected range. Based on the png tool posted by @JinghangLi, I expect the range to be between -150 to 150. Here are some example locations from spawn_points:

```
Transform(Location(x=586.908142, y=-24.062836, z=0.300000), Rotation(pitch=0.000000, yaw=-179.580566, roll=0.000000))
Transform(Location(x=570.894653, y=-10.179686, z=0.300000), Rotation(pitch=0.000000, yaw=-179.580566, roll=0.000000))
Transform(Location(x=568.920288, y=-13.694233, z=0.300000), Rotation(pitch=0.000000, yaw=-179.580566, roll=0.000000))
Transform(Location(x=570.945984, y=-17.179491, z=0.300000), Rotation(pitch=0.000000, yaw=-179.580566, roll=0.000000))
Transform(Location(x=122.573395, y=52.258755, z=0.300000), Rotation(pitch=0.000000, yaw=-0.053955, roll=0.000000))
Transform(Location(x=120.270103, y=48.760906, z=0.300000), Rotation(pitch=0.000000, yaw=-0.053955, roll=0.000000))
Transform(Location(x=122.566803, y=45.258755, z=0.300000), Rotation(pitch=0.000000, yaw=-0.053955, roll=0.000000))
Transform(Location(x=120.263512, y=41.760914, z=0.300000), Rotation(pitch=0.000000, yaw=-0.053955, roll=0.000000))
```
Why are these spawn points not in the png's range? How can I know the location of these spawn_points?

blankdean, 2023-05-28:
@zeid Thanks so much! This is very helpful so I appreciate it. I opened the Town6 file using the tool you provided and see the image below. Based on the map, I would like to spawn cars on the road within the range of x(100, 600) and y(-255,-234). I highlighted the road in the image. However, when I call `get_spawn_points()`, I don't see any (x,y) value pairs that fall in this range? Does `get_spawn_points()` provide all possible spawn points on the map? I would like to use the list of coordinates provided by this function to place up to 64 vehicles on the road highlighted.

I imagine 2 potential reasons for my issue:

1. The Town6 xodr map isn't the same Town6 I am using from Carla since I don't see spawn points from `get_spawn_points()` matching the highlighted road from the map. In fact, I see spawn points from `get_spawn_points()` that aren't anywhere on the map. Below are some spawn points I see after calling the code below. You'll notice the y coordinates of 225-251 don't even exist on the xodr map - the max y coordinate is about 155.

```
spawn_points = current_map.get_spawn_points()
for point in spawn_points:
    print(point)
```
```
Transform(Location(x=8.488513, y=225.960938, z=0.300000), Rotation(pitch=0.000000, yaw=-90.903801, roll=0.000000))
Transform(Location(x=5.002755, y=227.516144, z=0.300000), Rotation(pitch=0.000000, yaw=-90.903801, roll=0.000000))
Transform(Location(x=12.001884, y=227.405716, z=0.300000), Rotation(pitch=0.000000, yaw=-90.903801, roll=0.000000))
Transform(Location(x=22.377558, y=251.533249, z=0.300000), Rotation(pitch=0.000000, yaw=0.019546, roll=0.000000))
```
2. The `get_spawn_points()` function isn't providing all spawn points.

**Do you have any experience with this issue or can provide any suggestions? Thanks again.**

[image]

JinghangLi, 2023-05-28:
@blankdean Hi, In my understanding, Carla's spawn_points seem to be sparsely arranged on the map. So, If you are trying to have the vehicle generated at a specific location, I have a not particularly elegant solution that might help you. The sample code is as follows：
```
# Want to generate car at coordinate (x=1, y=1, z=0)
location = carla.Location(x=1, y=1, z=0)

# Get the nearest waypoint to the coordinate
client = carla.Client('localhost', 2000)
client.set_timeout(10)
world = client.load_world('Town06')
current_map = world.get_map()
waypoint = current_map.get_waypoint(location)

# Get our own spawn_point, set z=0.3 to avoid Z-collisions
waypoint_loc = waypoint.transform.location
spawn_point = carla.Transform(carla.Location(x=waypoint_loc.x , y=waypoint_loc.y , z=0.3),
                                                       waypoint.transform.rotation)
vehicle = world.spawn_actor(blueprint, spawn_point)
```

In fact, we can generate vehicles on any spawn_point = carla.Transform() that we define.

The reason for using waypoints is that when generating vehicles we want the x-direction of the vehicle to be parallel to the lane line, so we need waypoint.transform.rotation, also, waypoints are densely distributed on the map.

Hope this will help you.

blankdean, 2023-05-31:
Thanks @JinghangLi! This is an interesting method. I tried using `get_waypoint()`, but it sadly does not generate locations that are close to the ones I provide. **Is the xodr map an actual representation of the available x,y coordinates?**  @zeidk 

Here is something I tried:

```
x_range=(100, 400)
y_range=(-255, -234)
for _ in range(num_cars):
        x = random.uniform(x_range[0], x_range[1])
        y = random.uniform(y_range[0], y_range[1])

        print("Attempting to place vehicle at", x, y)
        location = carla.Location(x=x, y=y)
        
        # Get the nearest waypoint to the location
        waypoint = current_map.get_waypoint(location)

        waypoint_loc = waypoint.transform.location
        print("Waypoint_loc: ", waypoint_loc.x, waypoint_loc.y)
```

My terminal output:
```
Enter the number of vehicles to generate: 2
Attempting to place vehicle at 264.7850494217064 -254.474774140324
Waypoint_loc:  265.62481689453125 -23.904403686523438
Attempting to place vehicle at 128.1359943884197 -250.3125744988747
Waypoint_loc:  12.849374771118164 -146.81350708007812
```

From the output, you can tell the locations generated from `current_map.get_waypoint(location)` are not in close to proximity to the coordinates I provide. I am using an x and y range given by the map on [the odrviewer website](https://odrviewer.io/). **Is there something I'm missing? I'm having difficulty finding a legitimate map of Town 6 - I think this is my issue. Any additional insight or suggestions are greatly appreciated!**

Silenture, 2024-01-15:
The method provided by zeidk is useful: 
Clone this repository which contains Town01-Town07 xodr files: https://github.com/carla-simulator/opendrive-test-files.git
Go to https://odrviewer.io/
Open one of the xodr files from the repo you cloned (bottom left button)
Mouse hover the map and you should see the coordinates of your mouse's cursor in the top-right corner (see attached picture).
Use these coordinates to spawn your vehicle.

But there is a problem:
When you open town06.xodr, the coordinates you see do not completely correspond to the coordinates of the spawn point you want in Carla. The X-axis coordinates correspond and the Y-axis coordinates are opposite numbers. This can explain the problem encountered by blankdean. 
Give an example: If you want to generate a vehicle at a point (80, -150) on the town06 map opened in https://odrviewer.io/, the actual spawn point should be (80, 150).

liloongjia, 2024-01-15:
The method provided by zeidk is useful: 
Clone this repository which contains Town01-Town07 xodr files: https://github.com/carla-simulator/opendrive-test-files.git
Go to https://odrviewer.io/
Open one of the xodr files from the repo you cloned (bottom left button)
Mouse hover the map and you should see the coordinates of your mouse's cursor in the top-right corner (see attached picture).
Use these coordinates to spawn your vehicle.

But there is a problem:
When you open town06.xodr, the coordinates you see do not completely correspond to the coordinates of the spawn point you want in Carla. The X-axis coordinates correspond and the Y-axis coordinates are opposite numbers. This can explain the problem encountered by blankdean. 
Give an example: If you want to generate a vehicle at a point (80, -150) on the town06 map opened in https://odrviewer.io/, the actual spawn point should be (80, 150).