carla-simulator/carla #1934: how to access rgb camera data in client python code?

chankim, 2019-07-25:
how should I access rgb camera data in client python code? For example, if I want to run image recognition for the rgb data captured from the camera? Is there an example?
Thanks!

varunjammula, 2019-07-25:
https://carla.readthedocs.io/en/stable/connecting_the_client/

chankim, 2019-07-25:
@varunjammula hi, thanks. but I can't find the client_example.py. I tried to give manual_control.py "--images-to-disk" option but it gave me "manual_control.py: error: unrecognized arguments: --images-to-disk". error. Is this option a new one added in 0.9.6? I'm trying to acess the image data while the client is running, not anaylzying the image after the simulation has finished. But I could find it out if this --images-to-disk option works. I'll try accessing the pygame.Surface data. (the HUD is displaying the image on the pygame.Surface object)

chankim, 2019-07-25:
in my debug windown, I tried this 
(Pydb) dir(world.camera_manager.surface)
['__class__', '__copy__', '__delattr__', '__doc__', '__format__', '__getattribute__', '__hash__', '__init__', '__new__', '__reduce__', '__reduce_ex__', '__repr__', '__setattr__', '__sizeof__', '__str__', '__subclasshook__', '_pixels_address', 'blit', 'blits', 'convert', 'convert_alpha', 'copy', 'fill', 'get_abs_offset', 'get_abs_parent', 'get_alpha', 'get_at', 'get_at_mapped', 'get_bitsize', 'get_bounding_rect', 'get_buffer', 'get_bytesize', 'get_clip', 'get_colorkey', 'get_flags', 'get_height', 'get_locked', 'get_locks', 'get_losses', 'get_masks', 'get_offset', 'get_palette', 'get_palette_at', 'get_parent', 'get_pitch', 'get_rect', 'get_shifts', 'get_size', 'get_view', 'get_width', 'lock', 'map_rgb', 'mustlock', 'scroll', 'set_alpha', 'set_at', 'set_clip', 'set_colorkey', 'set_masks', 'set_palette', 'set_palette_at', 'set_shifts', 'subsurface', 'unlock', 'unmap_rgb']
maybe if I could use a method of this world.camera_manager.surface,..
I'll try later.

varunjammula, 2019-07-28:
What is the type(img) or typ(imgdata)?

cassfalg, 2019-08-02:
Take a look at `PythonAPI/examples/tutorial.py`. I found it to be a good starting point to extract some images. Not as complex code wise as manual_control.py or so.

chankim, 2019-08-03:
@cassfalg how would you write a code so that my ego-vehicle just don't move when it sees another vechicle? For example I want to access the bounding box sensor result. The tutorial.py doesn't give any clue regarding the usage of the sensor data. (I've seen how to set the control and apply it to my car)

chankim, 2019-08-03:
@varunjammula sorry, on second observation, for my command
 img = world.camera_manager.get_image() 
I got
AttributeError:"'CameraManager' object has no attribute 'get_image'"
and for my command
imgdata = pygame.surfarray.array_3d(img)
I got
AttributeError:"'module' object has no attribute 'array_3d'"

varunjammula, 2019-08-04:
in manual_control.py in CameraManager class, 
`_parse_image(weak_self, image)` function, the image is accessed in else loop as an array line 764-767.

closed 2019-10-10 by stale[bot]