carla-simulator/carla #1936: Get maximum steering angle of a vehicle

nikonikolov, 2019-07-25:
How can I get the maximum steering angle of a vehicle? The documentation says I can get it from the vehicle blueprint, but I don't see any such information. Please provide the exact way I can use to get the angle. Thanks!

I am using CARLA 0.9.5

manishthani, 2019-08-02:
Hi @nikonikolov ,
You could get the VehiclePhysicsControl of a vehicle and, in order to get the maximum steer angle for each wheel, you could get the `max_steer_angle` value from each `WheelPhysicsControl`. To sum up, you could do something like this:
```py
# Get Vehicle Physics Control
physics_control = vehicle.get_physics_control()

# For each Wheel Physics Control, print maximum steer angle
for wheel in physics_control.wheels:
    print (wheel.max_steer_angle)
```

ThrashAbaddon, 2019-08-23:
@nikonikolov can this issue be closed?

nikonikolov, 2019-08-23:
Thanks, this works.

closed 2019-08-23 by nikonikolov