carla-simulator/carla #1939: Current wheel angle/steering

pawel-ziecina, 2019-07-26:
Hi,
There is delay between applied steering and real steering - can be configured with 'Steering Input Rate' in vehicle blueprint. Is there possibility to read current wheel angle/steer?

Obtaining from vehicle.get_control() provides last VehicleControl command not current state (as stated in documentation).

CARLA version: 0.9.6

sheelabhadra, 2019-09-18:
Hi @pawel-ziecina, were you able to figure out a way to obtain the current wheel angle?

wulaali, 2019-10-18:
were you able to figure out a way to obtain the current wheel angle?

MohamedDaoud1, 2019-10-30:
If you are using carla ros bridge, I think it is the "steer" value in "vehicle_status" topic but normalized (max is 1)

closed 2020-01-05 by stale[bot]

mortenfc, 2020-04-16:
I need an answer for this. Bump

mortenfc, 2020-04-16:
> If you are using carla ros bridge, I think it is the "steer" value in "vehicle_status" topic but normalized (max is 1)

I don't think so, as the ros bridge just publishes the carla control received from the Python API.

sshawn9, 2025-01-02:
Actually, there is a python api called get_wheel_steer_angle which you can found [here](https://carla.readthedocs.io/en/latest/python_api/#getters_20), same for the c++ api. While it sucks as it will call the simulator and retrieve date when next tick, you may need single thread to continuously get current steer angle. I hope Carla team could more on vehicle kinematics and dynamics.

sshawn9, 2025-01-02:
> > If you are using carla ros bridge, I think it is the "steer" value in "vehicle_status" topic but normalized (max is 1)
> 
> I don't think so, as the ros bridge just publishes the carla control received from the Python API.

Yes, it just publishes the carla control received last time, ridiculous!