carla-simulator/carla #1941: Unreal Engine not detected

rebazica, 2019-07-29:
I'm trying to build Carla 0.9.3 on windows 10, with VS2017 and UE 4.21.2, however after commanding "make launch" on the command prompt x64, I eventually end up with the following error (last two lines):

ERROR: The system was unable to find the specified registry key or value.

-[BuildCarlaUE4]: [ERROR] Unreal Engine 4.21 not detected

Does anybody have any idea where I messed up?

closed 2019-10-04 by stale[bot]

michaelBfox, 2019-11-01:
Had the same Problem with UE 4.22
I solved it by creating the registry key manually. Maybe this is just a temporary workaround...

mikolololoay, 2019-11-21:
> Had the same Problem with UE 4.22
> I solved it by creating the registry key manually. Maybe this is just a temporary workaround...

Could you please explain to me how to create that key and how should it look? I have exactly the same problem right now.

seendsouza, 2020-01-05:
@mikolololoay Go to: HKEY_LOCAL_MACHINE\SOFTWARE\EpicGames\Unreal Engine\%UE_VERSION% (replace %UE_VERSION% with 4.22) and add a new String value with Value name `InstalledDirectory` and Value data your path to UE_4.22 (for me it is D:\Epic Games\UE_4.22

SuhrudhSarathy, 2020-05-07:
> @mikolololoay Go to: HKEY_LOCAL_MACHINE\SOFTWARE\EpicGames\Unreal Engine%UE_VERSION% (replace %UE_VERSION% with 4.22) and add a new String value with Value name `InstalledDirectory` and Value data your path to UE_4.22 (for me it is D:\Epic Games\UE_4.22

I am still getting this error after doing the above mentioned. Could you please help?

oeichenwei, 2020-08-13:
```
if not defined UE4_ROOT (
    set KEY_NAME="HKEY_LOCAL_MACHINE\SOFTWARE\EpicGames\Unreal Engine"
    set VALUE_NAME=InstalledDirectory
    for /f "usebackq tokens=1,2,*" %%A in (`reg query !KEY_NAME! /s /reg:64`) do (
        if "%%A" == "!VALUE_NAME!" (
            set UE4_ROOT=%%C
        )
    )
    if not defined UE4_ROOT goto error_unreal_no_found
)
```
So the there is not %UE_VERSION% in the registry can  work.

winterwindwang, 2021-07-12:
@SuhrudhSarathy Reboot your system maybe help to solve your problem.