carla-simulator/carla #1942: How to activate the "fixed time-step" mode if I use "make launch" to launch simulator?

1989Ryan, 2019-07-29:
Hi all. I am confused that how to use **fixed time step** mode if I use "make launch" to launch editor and run carla simulator?

bernatx, 2019-07-29:
Hi,

Just use 'make launch' to run the Carla server. Then go to the folder 'PythonAPI/util' and run the script 'config.py' to see the options you can change related to the frame rate. They are like these:
* **delta-seconds S**: set fixed delta seconds, zero for variable frame rate
* **fps N**: set fixed FPS, zero for variable FPS

You can choose to use one or the other. They are related as: **delta-seconds = 1 / fps**

So, using **config.py --fps 10** you will use a fixed time step of 10 frames per second.

1989Ryan, 2019-07-29:
> Hi,
>
> Just use 'make launch' to run the Carla server. Then go to the folder 'PythonAPI/util' and run the script 'config.py' to see the options you can change related to the frame rate. They are like [...]

Hi, 

Thanks for your reply. However, I am currently using version 0.9.4, and there are no such file under the folder `/util`. Then I copy the file `config.py` in the up-to-date repo in master branch but it doesn't work.

Dola47, 2019-08-06:
@1989Ryan  Hello Ryan, if you are using Carla 0.9.4 you can adjust your time step from the terminal before running Calra.

So, add to your command for example -fps=20, then you have timestep of 50 ms.

closed 2019-08-16 by 1989Ryan