carla-simulator/carla #1943: Unable to open any other map except Town01

strider175, 2019-07-29:
Hello, I recently updated Carla simulator to 0.9.6 version. When I run simulator like I used in previous versions I use command `./CarlaUE4.sh /Game/Carla/Maps/Town07 -windowed -ResX=320 -ResY=240 -benchmark -fps=30`. Whether I type 1 or 2 or 7 or any other Town it always opens Town01, which I assume is default map. So my question is if I am using wrong path or it could be that something was not installed.

Thanks in advance.

lijecaru, 2019-07-30:
Same as #1908
According to @jackbart94, the map can no longer be set at launch time. Instead, he says:

You now have to launch

./CarlaUE4.sh

then on another terminal go to PythonAPI/util and launch

python config.py -h

and it will give you the list of everything you can change, but now at runtime. For example, to put Town05 do

python config.py -m Town05

oldtopos, 2019-07-30:
This is annoying and complicated.  Have the other python examples been updated to allow these settings via command line ?  Looking at manual_control, the answer is no ...

Why is config.py not packaged so that it can be imported into the other tools ?

closed 2019-10-05 by stale[bot]

varad-more, 2020-06-03:
I'm facing same issue with current version 0.9.9.1