carla-simulator/carla #1946: How is the vehicle driving automatically?

qixiaoshuai0120, 2019-07-30:
Hi， 
   How is the vehicle driving automatically?How to use the XODR file generated from the Roadrunner software, I want to find the relevant function, thank you.

jackbart94, 2019-07-30:
I am stealing this answer that @marcgpuig gave on Discord:

"[the autonomous model in server side] is a really simple one and it's generated from an opendrive file. On the start of each (opendrive) road, we place a trigger box that contains an array of positions along the road. When a car hits one of these trigger boxes, it starts following these points in order, until it collides with another trigger box. For the traffic lights and stops, we have more triggers that just stop the car until something in the world changes the state of the trigger box, then the car can continue. All the cars have a set of raycast pointing the direction that they are going, these will tell the car if it has another vehicle in front (for cars stopping because other cars are stopped by, for instance, a traffic light)."

xiaoyang666, 2019-07-30:
Your information is very useful, then I try to find out where trigger boxes are, thank you.

qixiaoshuai0120, 2019-07-30:
Your information is very useful, then I try to find out where trigger boxes are, thank you.

closed 2019-10-05 by stale[bot]