carla-simulator/carla #1952: obstacle sensor distance problem

kimiya66, 2019-07-31:
Dear Carla Developers, @bernatx  @nsubiron @felipecode @TheNihilisticRobot ..
Some friends have checked the codes of ObstacleSensor.cpp, the problem that this sensor is not showing the correct distance to detected obstacle is that, instead of distance, it brings Radius values. and you could see the related line of code here in this link: https://github.com/irgendwie/carla/commit/ae6ba4de83e031c65ea96d773c6e223f93752b3b
After we changed this to the code and re-compiled Carla, it is fortunately working correctly. However, I want to ask you is it possible that you correct this code and re-compile Carla again or publish a new built release base on this?
because I have written all  of of my codes and run it in a released version of carla9.5 before where everything is working in my project except the distance of obstacleSensor.(I was running through ./CarlaUe4.sh or compiled version) but when I made this changes and run Carla from source code(by opening in Unreal Editor), every thing in my project disturbed, I mean the quality of map, Fps that my project worked with was 20Fps but in this way(run from source code and by opening UnrealEditor) I could not define sample rate and it changes so the codes are not working in new sample-rate, coordination also changed here compared to the time I was running my program with ./CarlaUE4.sh!!! (actors or vehicles coordinate that I set before)
That is why I need a compiled version again or do you know why every thing that I set before is changing when I run it in this way(with UnrealEditor issue#22 or from source code)?

Thank you for your help,

nsubiron, 2019-08-20:
You could open a [pull request](https://help.github.com/en/articles/creating-a-pull-request) with the change, that way it'd be easier for the developers to discuss and commit the fix :wink:

cassfalg, 2019-08-22:
@kimiya66 you can run `make package` to compile your own binary. You'll have a `CarlaUE4.sh` in the Dist folder that you can run same as the binary version.

kimiya66, 2019-09-03:
> @kimiya66 you can run `make package` to compile your own binary. You'll have a `CarlaUE4.sh` in the Dist folder that you can run same as the binary version.

Thank you it was really helpful :)

kimiya66, 2019-09-03:
> You could open a [pull request](https://help.github.com/en/articles/creating-a-pull-request) with the change, that way it'd be easier for the developers to discuss and commit the fix 😉

yes I will :)

PR #2182 (2019-12-10): Return HitDistance instead of HitRadius

#### Description

Fixes #1952 

#### Where has this been tested?

  * **Platform(s):** ...
  * **Python version(s):** ...
  * **Unreal Engine version(s):** ...

#### Possible Drawbacks

---
This change is [[image]](https://reviewable.io/reviews/carla-simulator/carla/2182)
closed 2019-11-09 by stale[bot]