carla-simulator/carla #1953: Recommended Ubuntu and Python versions

elishafer, 2019-07-31:
Anyone know what the recommended ubuntu/python versions are for Carla. I'm running Ubuntu 18.04 with python3.6 with precompiled Carla 0.9.5. I think that I may be getting sync errors because of that and maybe some other bugs as well.
I tried compiling on ubuntu 18.04 and I'm getting into dependency hell =[

jackbart94, 2019-08-01:
What I suggest is:

Ubuntu 16.04
Unreal Engine 4.22
Python 3.5.2 and 2.7.12

elishafer, 2019-08-01:
Thanks, so I take that this is what the precompiled versions in [releases](https://github.com/carla-simulator/carla/releases) were built for?

jackbart94, 2019-08-01:
Releases are built for Ubuntu 16.04 and after. If you carefully read [here](https://carla.readthedocs.io/en/latest/how_to_build_on_linux/) it's specified that it requires minimum Ubuntu 16.04, but it works for the latest distributions as well. Let me know if you manage to fix your issue by using these versions, but you should be able to build CARLA in Ubuntu 18.04 anyways.

closed 2019-10-07 by stale[bot]