carla-simulator/carla #1955: make PythonAPI error on Windows

erwef, 2019-07-31:
I have been trying to compile Carla for Windows 10 for a few days now but end up getting stuck on errors. Currently, I am trying to make PythonAPI but end up with the following error:

-[BuildPythonAPI]: [Batch params]: --py3
INFO: Could not find files for the given pattern(s).

-[BuildPythonAPI]: [ERROR] An error ocurred while executing the py.
-[BuildPythonAPI]: [ERROR] Possible causes:
-[BuildPythonAPI]: [ERROR]  - Make sure "py" is installed.
-[BuildPythonAPI]: [ERROR]  - Make sure it is available on your Windows "py".
make: *** [pythonAPI] Error 1

I am trying to install Carla 0.9.4 running:
Windows 10
Python 3.5.0
Unreal Engine 4.22
Visual Studio 2017 15.9.14
py 1.8.0

Any help or advice with this issue and for building Carla on Win10 in general would be appreciated.

bernatx, 2019-08-01:
Hi. Just be sure that you use the "x64 Native Tools Command Prompt for VS 2017" to execute the make command, otherwise, you will get several errors about not finding files.

erwef, 2019-08-06:
I was already using the "x64 Native Tools Command Prompt for VS 2017" for executing the make commands, and could execute the other make commands except for make pythonapi.

I could not find a way to compile the pythonapi, and found several others who could make the other tools for carla but not the pythonapi on windows, therefore I ended up downloading the [compiled 0.9.5 version](https://github.com/carla-simulator/carla/releases/tag/0.9.5) instead. This seems to work so far.

closed 2019-08-06 by erwef

ravivaghasiya1998, 2023-09-07:
Hello 
I am trying to install carla 0.0.13 from source on windows.

it gives me the same error

........
..
..
...
  Circuit.cpp
  Element.cpp
  Node.cpp
  osm2odr.vcxproj -> C:\carla\Build\osm2odr-visualstudio\src\Release\osm2odr.lib
  Building Custom Rule C:/carla/Build/om2odr-source/CMakeLists.txt
  -- Install configuration: "Release"
  -- Installing: C:/carla/PythonAPI/carla/dependencies/lib/osm2odr.lib
  -- Installing: C:/carla/PythonAPI/carla/dependencies/include/OSM2ODR.h
-[BuildOSM2ODR]: OSM2ODR has been successfully installed in "C:\carla\PythonAPI\carla\dependencies\"
-[BuildPythonAPI]: [Batch params]: --py3
INFO: Could not find files for the given pattern(s).

-[BuildPythonAPI]: [ERROR] An error ocurred while executing the py.
-[BuildPythonAPI]: [ERROR] Possible causes:
-[BuildPythonAPI]: [ERROR]  - Make sure "py" is installed.
-[BuildPythonAPI]: [ERROR]  - py = python launcher. This utility is bundled with Python installation but not installed by default.
-[BuildPythonAPI]: [ERROR]  - Make sure it is available on your Windows "py".
make: *** [PythonAPI] Error 1