carla-simulator/carla #1967: Problems with image blur in Carla 0.9.6

cstamatiadis, 2019-08-02:
Hello everybody, 

I'm running Carla on Ubuntu 18.04. Until a couple of days ago I was using version 0.9.4, however, since I'm mostly working on object detection, I decided to switch to 0.9.6, given the increased variety of pedestrians. (I was and am working with the pre-built projects). 
When I store the images they seem to be quite blurred compared to how they used to be. I added the following to my RGB sensor configuration: 
`motion_blur_intensity = 0

motion_blur_max_distortion = 0,`

but it doesn't seem to change anything. The pedestrian in the image below is quite blurred in my opinion. But also images in general seem more blurry.

[image]
[image]

For reference this is a sample of how images used to look before
[image]

Does anyone have an idea of what I could be doing wrong?

kimiya66, 2019-08-06:
I had also the same problem with new version so I try not to migrate to new version and apply all of the improvement on codes(like sensor problems...) manually to the source code of carla9.5 and compile it myself.

closed 2019-10-12 by stale[bot]

joao-paulo-alves, 2022-04-22:
Having the same problem on CARLA version 0.9.13

wuhanstudio, 2023-10-31:
The same problem with CARLA v0.9.14.

alexx-km, 2023-11-28:
The same problem exists in CARLA v0.9.15.