carla-simulator/carla #1977: make launch  return error - cannot find a compatible Vulkan driver

Siqi-Wang47, 2019-08-07:
I am the new to carla simulator. I have followed the document to install carla on Ubuntu 18.04. When I run 'make launch', it occurred an error like 'cannot find a compatible Vulkan dirver <ICD>'. I  tried to run './UE4Editor -opengl4' under the folder ~/Engine/Binaries/Linux, but it still doesn't work. Anyone knows how to solve this issue? I really appreciate your help.
[image]

qingmingjie, 2019-08-08:
You can modify the file yourself. PixelReader.cpp(No need to change it at present) and BuildCarlaUE4.sh(Delete Vulkan-related, if it doesn't work, delete opengl-related). No thanks.

asharalam11, 2019-08-08:
You can always install Vulcan from http://linuxconfig.org/install-and-test-vulkan-on-linux

Siqi-Wang47, 2019-08-08:
@qingmingjie @asharalam11 Thank you guys! I have solved it.

cassfalg, 2019-08-09:
@Siqi-Wang47: If the issue is resolved, please use the comment and close function to close this issue.

closed 2019-08-13 by Siqi-Wang47

jgollenz, 2019-09-24:
@Siqi-Wang47 How did you resolve it exactly? I have the same issue.

ankurjainjob, 2019-10-24:
how did you solve this issue

jgollenz, 2019-10-24:
@ankurjainjob I resolved this by checking if I had the newest driver installed for my gpu. By default there was some generic driver, so I downloaded and installed the current nVidia driver. It worked fine afterwards (you need to restart then and check if the driver did actually update)

ankurjainjob, 2019-10-24:
Thanks, but did you worked on VMware, sir I am working on VMware.

cassfalg, 2019-10-24:
If you want this to run in a virtual machine you need to ensure that you have proper 3D acceleration with according drivers. I don't know what the current state of affairs is for that, you need to investigate. A heavy 3D application like carla is not the ideal candidate for a vm though. Best open your own issue, also.

ibster123, 2020-01-29:
Heres A way I found in order to avoid this issue and work with the Unreal Editor without messing around with graphic drivers. In this way the Vulkan requirement is skipped and the editor can start properly. Using the terminal reach the linux binaries folder( cd /UnrealEngine/Engine/Binaries/Linux ) and then use this code in order to launch the editor under ( Opengl4 ./UE4Editor -opengl4 ) or ( cd UnrealEngine/Engine/Binaries/Linux ) then type ( UE4Editor ). It should work, if  it doesnt go to this site https://linuxconfig.org/install-and-test-vulkan-on-linux

wqt2018, 2021-02-14:
if you have Intel HD Graphic Card，try this method 
$ sudo apt-get install libvulkan1 libvulkan1:i386 mesa-vulkan-drivers mesa-vulkan-drivers:i386

cristinayinfa, 2021-06-24:
I have the same issue, I have installed Vulkan, but that did not fix it. I am also working in VM ware with installing carla

gurkansahin28, 2021-11-07:
On Mint Mesa-Vulkan-Drivers can solve it. https://mesa3d.org/

MasoumehX, 2022-01-28:
I have the same problem and none of these solutions worked for me. I have a Debian with VGA controller: intel corporate device 591b and install a Vulkan driver but I got the same error again. Could you please someone tell me what else I can do?