carla-simulator/carla #1978: 'World' object has no attribute 'get_actor' (Carla 0.9.5)

FelippeRoza, 2019-08-07:
I tried to get an actor by using its id using `World.get_actor(id)`, as described in the PythonAPI documentation. It turns out that this method does not exist.

To be sure, I've printed out `dir(world)` and `get_actors` is there, but no `get_actor` can be found.

code:

```
client = carla.Client('localhost', 2000)
client.set_timeout(5.0)
actor_list = []
world = client.get_world()
some_id = 123
vehicle = world.get_actor(some_id)
```

ppjtzjb, 2019-09-10:
same there. Did you figure it out?

ppjtzjb, 2019-09-10:
It turns out to be a feature only available with 0.9.6

FelippeRoza, 2019-09-11:
As a workaround I am getting the actor by using the following code, which works in the version 0.9.5:

`vehicle = next((x for x in world.get_actors() if x.id == vehicle_id), None)`

closed 2019-09-11 by FelippeRoza