carla-simulator/carla #1979: Losing some frames when saving images

Dola47, 2019-08-07:
Dear all,

I am currently recording my scenario (NoSignalJunction) by pressing R with the help of manual_control . However, when I change the saved images to a video, I found that some images or frames is lost.

I am using fixed time step of 50ms, FPS = 20 HZ.

Is there is a way to solve this issue, I want to have all of the images saved precisely.

Thanks.

cassfalg, 2019-08-10:
From what I understand, that is what synchronized mode is for. It also seems to do exactly that in my tests with creating a machine learning dataset, where I had similar issues. See https://carla.readthedocs.io/en/latest/configuring_the_simulation/#synchronous-mode

closed 2019-10-16 by stale[bot]

antoniohernandezm, 2022-02-08:
Hello @Dola47 ,

Did you manage to solve the frame loss?
I'm making a dataset with carla and the same thing is happening to me, I'm losing frames, even using the synchronous mode. I don't know how to solve it.