carla-simulator/carla #1983: Integrate bounding boxes into actor snapshot

eduardohenriquearnold, 2019-08-07:
The _WorldSnapshot_ and _ActorSnapshot_ interfaces provide a nice way to obtain precise pose of agents in the simulation at specific timestamps. However we still lack the bounding box information. 

Although one could rely on the getting it from the current actor object, it is not possible to guarantee that the actor hasn't been destroyed in the mean time.

eduardohenriquearnold, 2019-08-18:
I am working on the code and am deciding on where to store the bounding box information.
Right now I am proposing to store it within the `VehicleData` struct inside the `TypeDependentState` inside the `ActorDynamicState`, as in:
```
  struct VehicleData {
    VehicleData() = default;

    PackedVehicleControl control;
    float speed_limit;
    rpc::TrafficLightState traffic_light_state;
    bool has_traffic_light;
    rpc::ActorId traffic_light_id;
    geom::BoundingBox bounding_box;
}
```

However this is causing some default constructor problems:
```
opt/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Sensor/WorldObserver.cpp:25:62: error: call to implicitly-deleted default constructor of 'carla::sensor::data::ActorDynamicState::TypeDependentState'
  carla::sensor::data::ActorDynamicState::TypeDependentState state;
                                                             ^
opt/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/include/carla/sensor/data/ActorDynamicState.h:126:27: note: default constructor of 'TypeDependentState' is implicitly deleted because variant field 'vehicle_data' has a non-trivial default constructor
      detail::VehicleData vehicle_data;
```

Should I create a constructor for `VehicleData` or should I get the bounding box object directly inside the `ActorDynamicState` class?

nsubiron, 2019-08-20:
The main problem I see with this approach is that you're sending unnecessary data, every frame for every actor (not only vehicles) to every client connected to the simulator; furthermore this data is static, the bounding boxes do not change at runtime. IMO this ActorDynamicState is already overused, I added it initially as a temporary solution for getting the vehicle control, but if everyone starts adding their data here I have the feeling this is going to become quickly a bottleneck. But I totally see your point, this synchronization is tricky, we need to keep data at minimum for performance, but at the same autocomplete so we don't need to do unnecessary server calls. It's difficult to draw the line. Anyway, that's my opinion, let's see what @carla-simulator/dev think.

I think a better approach would be caching the bounding boxes in the client-side, but you have to request it while the actor is still alive though, and I think there's even a bug that you could exploit for this :grin: if you request actors by id `world.get_actors([id, ...])` it doesn't check whether the actor is still alive as long as the actor was previously requested (actor info is cached during an episode); but I haven't tested this, I don't have a pc with Carla now.

A feature that was previously discussed and could help with this was adding spawn and destroy events to the world snapshot, that could contain actor info. The idea would be notifying the client when an actor was spawned or destroyed during a given frame, with this we can even add `on_spawn` and `on_destroy` callbacks in Python (I think this can be useful for a lot of people).

Regarding the compile error, it is because the members of a union should be trivially constructible (because unions won't call constructor/destructors) https://stackoverflow.com/a/19764531/5308925.

closed 2019-10-26 by stale[bot]