carla-simulator/carla #1986: Make pedestrain walk cross the road

Siqi-Wang47, 2019-08-09:
Hi all,

I am the new to Carla. Now my project requires making the pedestrians walk cross the roads and interact with vehicles. But how could I do that? 
I am using the 0.9.6. I have tried to run the spawn_npc.py and set the destination to what I want in the map, but it doesn't work. Is there any way to make them go cross street? 
Thanks in advance.

ndeshp2s, 2019-08-12:
Hi @Siqi-Wang47,

Did you find any solution to this problem? I need to simulate a similar behaviour where I want pedestrians to cross the road.

Siqi-Wang47, 2019-08-13:
Hi @ndeshp2s ,

Unfortunately, I haven't got any solution to that. I think to solve this problem, we need to modify the map in which pedestrians could navigate on roads. You could try the method mentioned in #1899. I have tried that, but I couldn't install the Recast&Detour library.

ndeshp2s, 2019-08-15:
Hi @Siqi-Wang47,

Thanks for the information. I was wondering if it could be possible to use the older version of Carla where is functionality was available?

closed 2019-10-21 by stale[bot]