carla-simulator/carla #1987: [0.9.6] How to draw a map in which the pedestrian can navigate

Siqi-Wang47, 2019-08-09:
Hi Carla team!

I am very new to Carla and my project requires pedestrian walking on roads, but the current pedestrian blueprint cannot realize this idea. So I am thinking about redraw a  map to do this. I really want to know how you make the map available for pedestrians to navigate. Thanks in advance!

Siqi-Wang47, 2019-08-14:
Hi @bernatx , could you give me some hints for this problem?

bernatx, 2019-09-02:
Hi Siqi-Wang47, 

current version only has a basic navigation system for pedestrians. The navigation is based on a binary file created from the Recast&Detour library with all the information about where the pedestrians can walk. 

You can check this issue where some instructions and experiences are given: https://github.com/carla-simulator/carla/issues/1899

 I hope it helps.

closed 2019-11-08 by stale[bot]