carla-simulator/carla #1993: PythonAPI: Actor has empty object_type attribute

cmpute, 2019-08-13:
Actor.attributes for Vehicles has a field "object_type".
In 0.9.5 "object_type" had a value, e.g. "car".
In 0.9.6 "object_type" is always empty.

Test:
I'm using Carla 0.9.6 and latest ros-bridge on master branch. System is Ubuntu 18.04 and ROS melodic.
When I print the properties of the actor at https://github.com/carla-simulator/ros-bridge/blob/master/carla_ros_bridge/src/carla_ros_bridge/vehicle.py#L48, I found that the object_type is null, therefore all the objects in the ROS message are tagged as CLASSIFICATION_UNKNOWN. Please fix this! Attached is the screenshot for printing.

[image]

d-hayashi, 2019-10-07:
You may have already solved the problem, but I found a solution.

1. Open CARLA with Unreal Engine 4
2. Open the following file in the content browser
     1. Content -> Carla -> Blueprints -> Vehicles -> VehicleFactory
3. Select `GenerateDefinition` in Functions
4. Select `Vehicles` in Local Variables
5. Enter the object type in the `Object Type` of each element of the array
6. Save the file

[image]

closed 2020-02-11 by stale[bot]

roshambo919, 2022-09-30:
Confirming that it looks to me like object type is always empty in both 0.9.10 and 0.9.13 binaries. 

```
  blueprints = world.get_blueprint_library().filter('vehicle.*')
  for bp in blueprints:
      print('\n\n', bp.tags)
      for attr in bp:
          print(attr)
```

[image]

roshambo919, 2022-09-30:
This is important because, while I can use the number of wheels to differentiate between cars and bikes/motorcycles, I cannot differentiate between bikes and motorcycles without using the precise name of the actor.

yunmo555, 2022-10-08:
> This is important because, while I can use the number of wheels to differentiate between cars and bikes/motorcycles, I cannot differentiate between bikes and motorcycles without using the precise name of the actor.

Do you have a good solution to differentiate between bikes and motorcycles?

roshambo919, 2022-10-08:
I just hard-coded in the names of motorcycles and assigned based on the complete actor name. Not ideal if Carla adds new actor types but fine for now.

yunmo555, 2022-10-09:
- 

> I just hard-coded in the names of motorcycles and assigned based on the complete actor name. Not ideal if Carla adds new actor types but fine for now.

ok, there seems to be no better way at the moment. I try to modify 'object_type' attributes of blueprint, but it's read-only. 

[image]