carla-simulator/carla #1994: RuntimeError: time-out of 10000ms

kraken24, 2019-08-13:
I am using carla 0.9.5 for reinforcement learning. I tried to run simulation for 10000 episodes and after about 2200 episodes, i get the following error

[image]

Can anyone help and explain why does simulator shutdown??

cassfalg, 2019-08-13:
This looks like a timeout from your client script, not the server. It is waiting for a reply from the server and not getting it for 10 seconds, which is the configured timeout, so you get the error. There are a number of reasons for this...
Is the server still running? If not please supply it's logs (as text, not screenshot).
Is the server simply busy? As in your simulation runs too slow. You may need to switch to synchronous mode.

kraken24, 2019-08-14:
Thank you @cassfalg 
As you requested, here is the crash log file.

[CarlaUE4.log](https://github.com/carla-simulator/carla/files/3501708/CarlaUE4.log)

1. Server stopped by itself
2. I am running simulations on synchronous mode
[image]
3. There is only one client connected to the server

cassfalg, 2019-08-19:
According to this code picture you're not setting up a time interval for sync mode, which you should according to the docs. Try adding this line to your settings:
```
settings.fixed_delta_seconds = 1.0 / TICKS_PER_SECOND
```

Not sure if that fixes this error though:
```
[2019.08.14-13.27.27:204][122]LogOutputDevice: Warning: 

Script Stack (0 frames):

[2019.08.14-13.27.27:235][122]LogCore: Error: appError called: Assertion failed: AllTireConfigs[TireConfigID] == this [File:/var/lib/jenkins/UnrealEngine_4.21/Engine/Plugins/Runtime/PhysXVehicles/Source/PhysXVehicles/Private/TireConfig.cpp] [Line: 89] 

[2019.08.14-13.27.27:329][122]LogCore: === Critical error: ===
Unhandled Exception: SIGSEGV: invalid attempt to write memory at address 0x0000000000000003

[2019.08.14-13.27.27:329][122]LogCore: Assertion failed: AllTireConfigs[TireConfigID] == this [File:/var/lib/jenkins/UnrealEngine_4.21/Engine/Plugins/Runtime/PhysXVehicles/Source/PhysXVehicles/Private/TireConfig.cpp] [Line: 89] 

0x00000000044416d2 CarlaUE4!_ZN20FGenericPlatformMisc14RaiseExceptionEj(+0x62)
0x000000000475fc3e CarlaUE4!_ZN22FUnixErrorOutputDevice9SerializeEPKDsN13ELogVerbosity4TypeERK5FName(+0x13d)
0x00000000045c408e CarlaUE4!_ZN13FOutputDevice8LogfImplEPKDsz(+0x18d)
0x00000000045521f7 CarlaUE4!_ZN6FDebug12AssertFailedEPKcS1_iPKDsz(+0x1d6)
0x0000000003fc59ee CarlaUE4!_ZN11UTireConfig12BeginDestroyEv(+0x19d)
0x000000000495f535 CarlaUE4!_ZN7UObject23ConditionalBeginDestroyEv(+0xc4)
0x00000000048f9f19 CarlaUE4!_Z24UnhashUnreachableObjectsbf(+0x298)
0x00000000048f86c5 CarlaUE4!_Z23IncrementalPurgeGarbagebf(+0x264)
0x0000000007087550 CarlaUE4!_ZN7UEngine25ConditionalCollectGarbageEv(+0x53f)
0x0000000006a1e843 CarlaUE4!_ZN6UWorld4TickE10ELevelTickf(+0x4092)
0x000000000688facf CarlaUE4!_ZN11UGameEngine4TickEfb(+0xbae)
0x0000000003ec6107 CarlaUE4!_ZN11FEngineLoop4TickEv(+0x29a6)
0x0000000003ecb82a CarlaUE4!_Z11GuardedMainPKDs(+0x279)
0x000000000789acdb CarlaUE4!_Z14CommonUnixMainiPPcPFiPKDsE(+0x95a)
0x00007f99ba8e6830 libc.so.6!__libc_start_main(+0xef)
0x0000000003ebd029 CarlaUE4!UnknownFunction(0x3cbd028)

[2019.08.14-13.27.27:339][122]LogExit: Executing StaticShutdownAfterError
```
I don't know what to make of it either, so if this error continues to crop up, then I can't help you further.

kraken24, 2019-08-19:
Thank you @cassfalg. I will try this.

Does TICKS_PER_SECOND mean frames per second?

kraken24, 2019-08-19:
Thank you @cassfalg .

I will try this

cassfalg, 2019-08-19:
Yes, I used a global constant with that name, you can just pick any number equal or greater than 10 (according to docs, 10 is the minimum that the physics engine needs, so it always uses at least 10).

kraken24, 2019-08-20:
Hello @cassfalg 
Can you explain what is Timeout for Block on Render fence?? Its value is set at 300000

It did not work and Carla shutdown again. 
I am using fixed time step in synchronous mode as you instructed but then why does frame rate drop to zero?

Is it because i set wheel friction to a new value??

Here is the screesnshot
[image]

cassfalg, 2019-08-20:
I think the red part is the error, not the part about the render fence timeout. And that seems to be about tires, so maybe. But I am just speculating what that could mean or how to help you, sorry.

closed 2019-10-26 by stale[bot]

blazejosinski, 2019-11-08:
I'm observing the same exception with carla 0.9.6.

```Assertion failed: AllTireConfigs[TireConfigID] == this [File:/data/carla/UnrealEngine_4.22/Engine/Plugins/Runtime/PhysXVehicles/Source/PhysXVehicles/Private/TireConfig.cpp] [Line: 89]```

This happens only from time to time, even though I'm running carla with the same config.

ZhangYouRong, 2020-03-12:
@kraken24 
hello kraken24,
have you solved this issue,
I met the same problem.

carla 0.9.5, python 3.7, win10

stavBodik, 2020-03-17:
same error here :

[2019.08.14-13.27.27:329][122]LogCore: Assertion failed: AllTireConfigs[TireConfigID] == this [File:/var/lib/jenkins/UnrealEngine_4.21/Engine/Plugins/Runtime/PhysXVehicles/Source/PhysXVehicles/Private/TireConfig.cpp] [Line: 89] 

can any one explain what is this TireConfigs ? maybe just need to comment out this assetion bug line ?

ZhangYouRong, 2020-03-17:
When I try my code on 0.9.5 version, this error disappear.But I cannot explain why?Maybe just a bug on 0.9.8?@stavBodik

stavBodik, 2020-03-17:
@ZhangYouRong ZhangYouRong

I also dont remmber I had this on 0.9.6

kraken24, 2020-04-01:
@ZhangYouRong 

I am sorry for late reply. I hope you were able to solve the problem.
I was using carla on linux.

The mistake I made was that i was recalling set function to set value of tire friction every iteration. Once I stopped doing that and only called it once before the start of simulation, the error was solve