carla-simulator/carla #1996: Support multiple "ticking" clients within synchronous mode

fpasch, 2019-08-14:
Within [synchronous-mode](https://carla.readthedocs.io/en/latest/configuring_the_simulation/#synchronous-mode) it is currently only supported to have exactly one CARLA-Client that calls [tick()](https://carla.readthedocs.io/en/latest/python_api/#carla.World.tick).

For more complex scenarios, (e.g. with multiple instances of an ad-stack-client) it should be supported to have multiple CARLA-clients that independently can call tick().

To achieve that, CARLA needs to wait for all clients executing tick() and proceed afterwards.

It needs to be discussed how to identify if a CARLA client wants to send ticks(). Either
1. in synchronous mode, all clients need to tick
2. CARLA clients need to flag, that they want to tick (e.g. during initialization)

I would vote for solution 2, as it would only affect "active" clients (i.e. that contain a feedback loop).

nsubiron, 2019-08-20:
This was discussed when we proposed 0.9.x API. Solution 2 sounds good, but I think it could be challenging to make it right and not end up with deadlocks when one of the clients disconnects or dies, specially at initialization. At the time we decided to have just one "master" client that ticks, the rest of the clients can communicate with this one if you need more complex behaviors, but this would have to be done by users in Python. (In fact, one of the things that were proposed was having a tick per actor, you could create "sync actors" and the simulator would wait until all the sync actors tick).

fpasch, 2019-08-21:
You're right. Within a single system setup, the "master"-client approach might be feasible.

But it gets messy if you have a multi machine setup. You would have to establish a new communication-pipeline in parallel to Carla to keep the clients in sync.

closed 2019-10-27 by stale[bot]

LucasFKobernic, 2022-02-03:
Hi @nsubiron 

"At the time we decided to have just one "master" client that ticks, the rest of the clients can communicate with this one if you need more complex behaviors"

I look in the documentation how to do that but i didn't find anything. Is this fearture already implemented in CARLA 0.9.12 or 0.9.13?