carla-simulator/carla #1998: Exporting map created in UE4 for compiled version.

ruichen-v, 2019-08-14:
My goal is to edit a map for use with the compiled version (0.9.4) of Carla. My new map can be loaded by the simulator. However, when I try to connect a client to Carla, it says `Fatal Python error: PyEval_SaveThread: NULL tstate` and crashed. I checked the terminal output of compiled Carla and found that it was due to a missing OpenDrive file **Content/Carla/Maps/OpenDrive/A.xodr**. Since I didn't use OpenDrive to create the map, how can I resolve this? (e.g. generate .xodr file from .umap and .uasset, bypass the file load, etc.?)

Here's what I've done:

1. Duplicate Town02, made some change and saved in another name A.
2. Put the **A.umap** and **A.uasset** file in `carla_root/Unreal/CarlaUE4/Content/Carla/ExportedMaps` and ran **carla_root/Util/BuildTools/ExportMaps.sh**.
3. Move the generated **ExportedMaps/A.tar.gz** to `carla_compiled_root/ExportedMaps` and ran **carla_compiled_root/ImportMaps.sh**.
4. Move **A.uexp** and **A.umap** from `carla_compiled_root/CarlaUE4/Content/Carla/ExportedMaps` to `Content/Carla/Maps/`. **A_BuiltData.uasset** and **A_BuiltData.uexp** are already in the `Maps/` folder.
5. Run `./CarlaUE4.sh /Game/Carla/Maps/A -windowed -ResX=1280 -ResY=720 -benchmark -fps=60`
6. Run `python manual_control_steeringwheel.py`

*Error in Carla terminal:*
```
[2019.08.14-19.00.19:065][202]LogTemp: Error: Failed to load OpenDrive file '../../../CarlaUE4/Content/Carla/Maps/OpenDrive/A.xodr'
```

*Error in Client terminal:*
```
$ python manual_control_steeringwheel.py
pygame 1.9.6
Hello from the pygame community. https://www.pygame.org/contribute.html
INFO: listening to server 127.0.0.1:2000

Welcome to CARLA manual control with steering wheel Logitech G29.

To drive start by preshing the brake pedal.
Change your wheel_config.ini according to your steering wheel.

To find out the values of your steering wheel use jstest-gtk in Ubuntu.

Fatal Python error: PyEval_SaveThread: NULL tstate
[1]    6588 abort (core dumped)  python manual_control_steeringwheel.py
```

matthew-rococo, 2019-08-15:
You just need copy another opendrive file and rename it to your map name(your map name.xodr). put it at the opendrive directory

获取 Outlook for Android<https://aka.ms/ghei36>

ruichen-v, 2019-08-17:
Thanks! It worked.

closed 2019-08-17 by ruichen-v

lilijun1130, 2021-08-20:
Hello, how to export map in carla0.9.11? I cannot fine ExportMaps.sh in carla0.9.11

ronyshaji, 2022-12-15:
> Hello, how to export map in carla0.9.11? I cannot fine ExportMaps.sh in carla0.9.11

Did you find a method to do this ?

Danlaw-Nikhil, 2024-01-22:
I have the .uasset file and .umap files that i got from UE4 Carla fork. but how do i export this as a tar.gz ?

xswings, 2024-12-28:
Excuse me, does anyone know how to import a renamed duplicate map in Carla 0.9.11 or later? I need to export a modified map from the source-built version in Unreal Engine 4 and integrate it into the compiled version of Carla.