carla-simulator/carla #1999: Add ability to delete debug elements

cmpute, 2019-08-15:
This is related to https://github.com/carla-simulator/ros-bridge/issues/173.

In order to implement Marker.MODIFY or Marker.DELETE operation, I wish to achieve the capability to edit or remove debug elements (arrows, boxes, etc) created by debug helpers. A possible way to do this is to get the 'actor' (?) id of the debug elements when creating and allow modify or deleting the elements by their id.

closed 2020-02-19 by stale[bot]

MasterIzumi, 2022-10-12:
Any solution for this kind of operation in 2022? :)

lesurp, 2022-12-07:
Still nothing about this? I can afford to spend some time on it but I have never looked at the carla internals so that might take a while :/

Youjin1985, 2025-03-25:
Bump... really need this, its extremely annoying to reload map after every simulation tjust to get rid of debug marks

astralmaster, 2025-04-18:
Bump. Having capability to draw debug elements also necessitates the functionality to erase them.

astralmaster, 2025-04-20:
Got tired of waiting, here's a PR:  https://github.com/carla-simulator/carla/pull/8855

cmpute, 2025-04-21:
> Got tired of waiting, here's a PR: [#8855](https://github.com/carla-simulator/carla/pull/8855)

You are a hero! It's ignored for nearly 6 years...

PR #8855 (2025-04-22): Added ability to clear debug shapes without reloading simulator UE4
#### Description

Adds the ability to clear debug shapes without having to reload the simulator. Using World.PersistentLineBatcher->Flush() .
In PythonAPI called by:

```
world = client.get_world()
debug_helper = world.debug
debug_helper.clear_debug_shape()
```

Fixes #1999 
  

#### Where has this been tested?

  * **Platform(s):** Windows 10 x64
  * **Python version(s):** 3.7.0
  * **Unreal Engine version(s):** 4.26.2

#### Possible Drawbacks

If `PersistentLineBatcher` is used elsewhere in the code apart from `FDebugShapeDr [... cut]