carla-simulator/carla #2000: How do I translate speed and acceleration into throttle and brake to control the car

JIXIMD, 2019-08-15:
VERSION:CARLA.0.95
OS:UBUNTU16.04

Hi, 
   Is there an API interface to support this transformation?
   If not, what should I do?
Thank you for your help

jan-reich, 2019-08-18:
No, there is not an official API, as this depends on quite some parameters of the vehicle. What you can do however is creating a set of PID controllers yourself or use the ROS node at https://github.com/carla-simulator/ros-bridge/tree/master/carla_ackermann_control that does the job. Note that you have to tune the controllers to the used vehicle!

wulaali, 2019-09-09:
> No, there is not an official API, as this depends on quite some parameters of the vehicle. What you can do however is creating a set of PID controllers yourself or use the ROS node at [...]

Is there any recommended PID parameters for vehicle mustang？

closed 2019-11-15 by stale[bot]

SaeedBASALEH, 2023-07-20:
Hi, 
For me it is not clear. I want to know if in CARLA 9.11 there is a way to convert acceleration to throttle. I saw different projects on car control using CARLA simulator, but almost all of them convert the acceleration value if it is positif to throttle taking in account the throttle is between 0 and 1, and same analogy for breaking.