carla-simulator/carla #2001: vehicle location

moein1924, 2019-08-15:
i would like to store a vehicle position and use it later on, but when I use vehicle.get_location().x, the value changes all the time, no matter where i put it. is there a way to record and save a location x, y of vehicle in a snapshot and use it later?

cassfalg, 2019-08-19:
Not really sure what you want do do or where your issue is exactly...

However, if you've got the vehicle object, there's the `get_transform()` and `set_transform` methods. Or you can just supply the saved transform to the `world.spawn_actor(...)` method. Depending what you want to do, the transforms are probably better than the location objects because they save the vehicle's location and orientation, not just the location.

closed 2019-10-25 by stale[bot]

PTorr, 2020-07-06:
get_location and get_transform returns values that change all the time.
It's advancing the car, why?