carla-simulator/carla #2002: problem

llove-y, 2019-08-15:
BuildLibCarla.sh: Success!
BuildCarlaUE4.sh: Using Unreal Engine at '/home/udi/UnrealEngine_4.19'
BuildCarlaUE4.sh: Build CarlaUE4 project.
make[1]: Entering directory '/home/udi/carla-0.9.5/Unreal/CarlaUE4'
mono "/home/udi/UnrealEngine_4.19/Engine/Binaries/DotNET/UnrealBuildTool.exe" CarlaUE4Editor Linux Development  -project="/home/udi/carla-0.9.5/Unreal/CarlaUE4/CarlaUE4.uproject" 
Using 'git status' to determine working set for adaptive non-unity build.
Creating makefile for CarlaUE4Editor (no existing makefile)
Performing full C++ include scan (no include cache file)
ERROR: Could not find definition for module 'ImageWriteQueue' (referenced via Target -> Carla.Build.cs)
Makefile:620: recipe for target 'CarlaUE4Editor' failed
make[1]: *** [CarlaUE4Editor] Error 5
make[1]: Leaving directory '/home/udi/carla-0.9.5/Unreal/CarlaUE4'
Util/BuildTools/Linux.mk:70: recipe for target 'CarlaUE4Editor' failed
make: *** [CarlaUE4Editor] Error 2

when I build the source code of carla 0.9.5 , I have some trouble like this;what I can do to fix it?

cassfalg, 2019-08-19:
Are you sure that Unreal Version 4.19 is the correct one for 0.9.5? Why don't you use the packaged version if you want to use 0.9.5? Why not use the current version if you want to compile it? Did you try to clean the build directories and try again? What OS is this on? Which part did you try to compile/run?

closed 2019-10-25 by stale[bot]

patilrajeevs, 2021-02-25:
I am facing this issue, was there a resolution @Ryoma-Liu  ?