carla-simulator/carla #2004: How to add custom vehicle in pre-compiled version?

asharalam11, 2019-08-16:
Hi, I have successfully added my custom vehicle in Carla Source Build (using UE4 Editor). However, I do not know how to use the Blueprints and assets I created and add them to Carla's precompiled version 0.9.6. Thanks for help

kbarbora, 2019-08-16:
As long as the assets created are similar to the default vehicles in carla, you should be able to drag them to the map.
if you need them to move (e.g. autopilot or something else) I have been trying to do so with no success.

asharalam11, 2019-08-16:
I don't think you can drag vehicles to map in the pre-compiled version. What I mean is that I don't have Unreal Engine installed in the pre-compiled version. Do I need to install Unreal Engine in pre-compiled version to add vehicles to the map?
I can copy assets I created in the source build to the pre-compiled version; but how do I add them to the vehicle factory in the Pre-compiled version?

kbarbora, 2019-08-16:
oh ok, I miss the 'pre-compiled version' part. You can do that using the python API really easy.

asharalam11, 2019-08-16:
Wait....how??? Can you please guide me?

kbarbora, 2019-08-22:
blueprints = world.get_blueprint_library().filter("vehicle.")  // all available vehicle models
blueprint = random.choice(blueprints) // choose one at random
transform = world.get_map().get_spawn_points()[0]
vehicle = world.spawn_actor(blueprint, transform)

asharalam11, 2019-08-22:
How do I modify this in an API; for example in manual_control.py????

In manual_control.py of **Source Build**; I modify vehicle.<custom blueprint> like below code and it works:
argparser.add_argument(
        '--filter',
        metavar='PATTERN',
        default='vehicle.tesla.*',
        help='actor filter (default: "vehicle.*")')

But when I do the same in pre-compiled version; I get the following error:
IndexError: Cannot choose from an Empty Sequence; even though I have copied the car I created in both Blueprints and Static->Vehicle folder of the Pre-compiled version.

Please help !!!

rmurrish, 2019-08-23:
> I have copied the car I created in both Blueprints and Static->Vehicle folder of the Pre-compiled version.

@asharalam11 how did you add it to the blueprints? I'm watching this issue closely, hoping it's possible without compiling from source.

asharalam11, 2019-08-23:
@rmurrish I wasn't successful in adding the blueprints in the pre-compiled version. I have merely copied it inside the folders. But I still can't spawn it; that is why I am asking the question

closed 2019-10-30 by stale[bot]

Joshevans22, 2020-04-09:
I am also experiencing the issue. Simply copying the assets from the editor to the pre-compiled version does not work. How do I add vehicles created in the editor to the pre-compiled version allowing them to be controlled via the python api?

Hopefully this is possible

Fmkunle, 2020-04-29:
Any luck on this so far ? I also have the same problem. (Using Vehicle factory or otherwise )

Any information would be really appreciated.

MS012086, 2021-01-20:
Dear All, 
I am also looking for same support, I have downloaded the truck from unreal engine market place, How to use it in pre-compiled version of Carla.

Thank you for your support in advance!

Thanks & Regards,
M Saravanan