carla-simulator/carla #2005: ./CarlaUE4.sh executable binary file not found

Cathe96, 2019-08-17:
Hello,

I am trying to run ./CarlaUE4.sh but receive the following error 
[image]

Could anyone possibly tell me on how I could fix the issue? 

Thanks

cassfalg, 2019-08-19:
What version of Carla is this? Binary download? Did you try to extract it again? Have you checked if the missing referenced file is there?

What Operating System? What shell? Which readlink command is in path? The one you're calling doesn't understand the -f option, so it's not the standard linux readlink command from ubuntu's coreutils package for example...

closed 2019-10-25 by stale[bot]

mainakchain, 2020-10-12:
I am also facing the same issue. My system is a Machintosh system. Please let me know what could be the possible issue. @cassfalg