carla-simulator/carla #2007: Radar support

ivannson, 2019-08-18:
As per the [roadmap](https://github.com/carla-simulator/carla/blob/master/Docs/img/carla_timeline.png) there were plans to add Radar support by the end of 2019 Q3. Judging from the versions associated with the new features on the roadmap, I understand that it will be later than that. Is there an ETA for the Radar addition?

ScottOGrady, 2019-09-04:
Same question as ivannson. I'm keenly looking forward for a radar sensor. Can you give some updates about your progress and some details / maybe a realistic timeframe when to expect it's release?
For my PhD project its an integral part of the simulation and I would be glad to get some info :)
Thanks in advance.

pprivulet, 2019-10-22:
I'm looking forward for a radar sensor too. Do any progress have been made ?

basavaraj1995, 2019-12-02:
Any progress have been made ?

basavaraj1995, 2020-01-14:
**CARLA 0.9.7 now supports RADAR, IMU and all the camera-based sensors are provided with an additional parametrized lens distortion shader.**

closed 2020-03-21 by stale[bot]